nodemcu/nodemcu-firmware #1101: Dead code pwm in firmware

TerryE, 2016-03-01:
This one was thrown up by  #1100.  At the moment the platform.h tier declares the platform abstraction for GPIO, SPI, UART, PMW, Sigma_delta, I2C and Flash which are all implemented in `platform.c`.  It also declares Eth and ADC which aren't implemented at all!   

The current implementation results in most of these functions being optimised out by the loader as unused if not referenced, however the pwm_init is dome as part of platform_init() and hence the some of the  PWM driver and platform_pwm code gets mapped in -- roughly 0.5Kb iram0 and 2Kb irom.  Likewise the gpio system also calls pwm close functions, so IMO its worth considering moving these lazily into the first routines that need them or suppress this logic if the PWM module is not enabled. Thoughts?

devsaurus, 2016-03-01:
My understanding of the module vs platform layer is
- app/modules are enabled or disabled per user_modules.h
- platform and drivers are linked as needed

Meaning that functionality of the Lua user interface is configured by these defines but the underlying routines are shared as needed. Theoretically, module A can use the platform/ driver routines which are exposed to Lua by module B even though module B is disabled.
This conclusion is just based on observation, but with this background I'd vote for

> moving these lazily into the first routines that need them

and not to exclude them just because Lua api is disabled.

pjsg, 2016-03-01:
Yeah -- I realized that the pwm stuff cannot be excluded -- and hence adding the init into pwms_init was the safe approach. pwms_init is called from somewhere pretty fundamental.

get_pin_map() should be called early (and then it should not re-run if it is called again). 

I was trying to make the minimal fix to get the system working again.....

TerryE, 2016-03-01:
We have the gcc flags `-Os -ffunction-sections -fdata-sections` together with the linker flag `--gc-sections` which means that platform and driver components that are used are retained and the unused one GCed without needing to code this logic up in the code.  So far so good.  The issue that I have is that `platform_init()` calls `pwm_init()` (but no initialisation for any of the other submodules such as platform_gpio).  This means that 3Kb of pmw components are linked into every firmware build, despite the fact that the vaste majority don't use PMW. 

Also `pwm_set_freq(500, 0)` and `pwm_start()` are execute on all builds.  In the former case the second channel parameter is ignored but the 500 is bounds checked (1,100) and sets `pmw.freq=500`.  As far as I can see, `pwm_start()` starts the H/W timer.  

The way that this code has unused variables commented out stuff and inline hard coding of settings, just smacks of in-progress development hacks, rather than code ready for release to production.  

> I realized that the pwm stuff cannot be excluded.

Philip, what is your rationale for this?  Why _must_ the pwm code be initialise on builds that don't use it (that is probably over 95%)?  I will enable the PWM debug and look at this a little more.  :smiley:

pjsg, 2016-03-01:
When I said that the PWM stuff cannot be excluded -- this was an observation on the current way that it is built. There is no reason (AFAIK) why it could not be brought into the same infrastructure as GPIO so that you need a LUA_USE_MODULES_PWM to make it be included. 

I can do this if you want.....

TerryE, 2016-03-01:
Actually as I said the whole of the PWM stuff has debris in it.  It could do with one of us going through it and giving it a houseclean.  If you are volunteering --  :boom: :sparkles: :+1: :clap:

pjsg, 2016-03-02:
Yes -- that would be my approach -- but you need to add the LUA_USE_MODULES_PWM to signal whether the code is included or not. I would default pwm as included (for backwards compatibility if nothing else).

I also realize that there are some nasty interactions between all the drivers that fiddle with pins -- there is no restriction that only one module touches a pin at a time. As a followup, I think it is worth having an arbiter record who owns which pins and prevents another module from taking a pin that is owned by someone else. The problem is that most of the modules don't have 'release/close' methods so there is no way to say that you are done with a pin. Anyway, all that is for another day....

pjsg, 2016-03-02:
I now have this code -- just needs combination testing.....   I wish there was a suite of tests that could be run....

TerryE, 2016-03-02:
>  I wish there was a suite of tests that could be run....

In my LCD patch (here is the [write-up](http://www.esp8266.com/wiki/doku.php?id=nodemcu-LCD)), I used the [Lua 5.1 test suite](http://www.lua.org/tests/) and found this a brilliant way exercising in not all then a _lot_ more code paths then I could have dreamt of.  Though to do this, I first had to implement the path in standard Lua and then once it had a 100% pass, port it to nodeMCU.  I did think at the time that it might be _really_ nice to have an equivalent suite for nodemcu, that could be downloaded onto a ESP-12E config, say and run from there, but it's on the wish list :disappointed:.

PR #1110 (2016-03-03): Omit all the PWM code if the PWM module is not included
This resolves #1101 -- it turned out pretty simple.

Also, I fixed the compiler warning that annoyed me in every build (the warning about unsigned < 0 is always true.)
closed 2016-03-03 by TerryE