nodemcu/nodemcu-firmware #1103: No COM port, no /dev/tty* - tried different OS's, different cables

piejanssens, 2016-03-01:
My devkit is an Amica V3, well two of them and both seem to have the same problem.
I tried to get the deivce to show up to then be able to install the nodemcu firmware.
I did this on different computers and different OS's (Windows 7/10 + OS X 10.11.4), everytime making usre I installed SiLabs CP210X drivers first.
No sign of the devkit anywhere.. When I unplug it and plug it back in the LED near the WiFi antenna blinks and then nothing. I pressed RST a few times, short/long, nothing.

I really hope you tell me that I'm stupid and that I should have RTFM so I would have not missed providing additional power to the board the first time you flash it... :smile: But I doubt that this is the case.

TerryE, 2016-03-01:
Pieter, sorry but this is wrong place to ask this sort of Q.  See #719 or our support page for relevant forum links.

closed 2016-03-01 by TerryE

piejanssens, 2016-03-01:
OK

Leaving this trace for those stumbling on the same issue: http://stackoverflow.com/questions/35727990/no-com-port-no-dev-tty-tried-different-oss-different-cables