nodemcu/nodemcu-firmware #1107: luac.cross compile is broken

TerryE, 2016-03-02:
This is a side-effect of the new use modules implementation. I am ashamed that we took so long to notice. I didn't because I've only recently rebuilt `luac.cross`.  Bugger.

TerryE, 2016-03-02:
For those that don't `luac.cross` (see #598) is a special build of the firmware that creates a cross-compile version of the standard Lua compiler command `luac`.  The main difference if that it is based on the nodemcu source code so the `lc` files can be copied to the ESP module for direct execution by the firmware.  To do this we need to address two issues:

1.  We only need the minimal compiler functionality and no runtime components so we can ignore all the modules, drivers, etc. and only need to use a subset of the `app/lua` files.  Some conditional code is needed  to dummy out hooks into the RTS.
2.  nodeMCU uses the newlib-based ELF, rather than Posix libraries, so the Lua files that are used for the cross compiler need a little preprocessor magic to switch in the right version for the cross version without breaking the standard build

I had sorted both (1) and (2) in my original PR, but what Johny and I unintentionally did in simplifying the module inclusion process is to break this dummying out, so now the `luac.cross` build fails.  So I will do a PR to fix this.  I will also add luac.cross to the documentation in this same PR.

jmattsson, 2016-03-02:
I haven't looked at the luac.cross, but the simple cross-compiler we're building from NodeMCU's Lua source at $work was not affected by this. Drop me an email if you want pointers for comparisons! We're effectively just using a small main() which wraps a luaL_loadstring() and luaU_dump_crosscompile()

TerryE, 2016-03-03:
See PR #1111.  Too late on the pointers :)  

One of the main advantages of `luac.cross` is that the full `luac` options including `-l` works.  Doing the following takes under 1 sec and throws up a couple of interesting errors
```bash
terry@nodemcu1:~/nodemcu-firmware$ (for f in $(find * -name "*.lua"); do ./luac.cross -l -o /dev/null $f; done) > /tmp/temp.lst
./luac.cross: examples/fragment.lua:400: '=' expected near '-'
./luac.cross: lua_examples/mqtt/mqtt_file.lua:46: malformed number near '192.168.18.88'
```

closed 2016-03-03 by TerryE