nodemcu/nodemcu-firmware #1108: Extremely strange and inconsistent behavior when accessing results of cjson.decode in function.

jaccarmac, 2016-03-02:
```
last_stamp = 0
function parseCmd(response)
   decoded = cjson.decode(response)
   sendCmd("stop")
   if last_stamp < decoded["timestamp"] then
      last_stamp = decoded["timestamp"]
      sendCmd(decoded["value"])
   end
end
```

Expected result: When a response with a greater timestamp is received, the code inside the `if` runs and `last_stamp` is updated.

Actual result: The code inside `if` gets run every time and `last_stamp` stays 0.

Note that the `if` condition works, and uses the value of `decoded["timestamp"]`. Only on assignment does the value start to behave strangely. `print` works on it, `type` of `decoded` is `table` and `type` of `decoded["timestamp"]` is `number`, which is expected.

I originally thought this might be a global variable issue, but the following case also behaves strangely, where the code inside the `if` is replaced.

```
last_stamp = 0
function parseCmd(response)
   decoded = cjson.decode(response)
   sendCmd("stop")
   if last_stamp < decoded["timestamp"] then
      local new_stamp = decoded["timestamp"]
      print(decoded["timestamp"])
      print(new_stamp)
   end
end
```

Inside the loop, the first `print` prints a number while the second prints `nil`.

I'm using an image generated on the website from master on 26 February. The configuration, according to the automated email, is `This was built against the master branch and includes the following modules: cjson, file, gpio, net, node, tmr, uart, wifi.` I am using the integer version of said firmware. If anyone wants an exact binary copy of it, I've [hosted the file](https://drive.google.com/file/d/0B_S86gZvPJWKdExDLUVWTW44SWM/view?usp=sharing) for perusal.

TerryE, 2016-03-02:
@jaccarmac Jaob, I am not sure whether you are reporting a specific bug within the firmware and in this case within the `cjson` module or whether is a request for help.  The former belongs on this list; for the latter see the forum links on our support page within the online documentation or #719.

For a bug I would expect you to post something of the form: 
```Lua
decoded = cjson.decode([[ ... some json encoded string ... ]])
print(decoded[some_field]) -- This says XXXX and should be YYYY
```
If you can't do this, then I will close this issue.

jaccarmac, 2016-03-02:
See my second snippet:

```lua
last_stamp = 0
function parseCmd(response)
   decoded = cjson.decode(response)
   if last_stamp < decoded["timestamp"] then
      local new_stamp = decoded["timestamp"]
      print(decoded["timestamp"])
      print(new_stamp)
   end
end
```

Specifically, I expect to see two lines printed, both which contain an identical value, a timestamp. Instead, the first line contains a timestamp and the second contains just `nil`.

Let me fire up the board and I can provide a screenshot of the behavior live.

jaccarmac, 2016-03-02:
[image]

There's an example of the behavior happening on-board. The lines at the top are from a timer which fires every few seconds and grabs data over HTTP, while what you see below is manual input to the misbehaving function in several formats. As you can see, the behavior doesn't depend on the format or source of the input string or the number of keys in the JSON document.

TerryE, 2016-03-02:
Read the error message char 3 is **v** not **"**.  This isn't  bug in cjson.

closed 2016-03-02 by TerryE

jaccarmac, 2016-03-02:
I mistyped the document once. Notice that when I quote the keys properly in the JSON I get the erroneous behavior described above  about 10 times.

Just to be 100%, absolutely, unambiguously clear, the code I have added to the ticket multiple times should allow assigning a variable from the decoded JSON string. However, while I can print a value directly from the decoded table, I cannot assign said value to a local or global variable. The `nil`s in the output above should not exist.

TerryE, 2016-03-03:
Sorry for that, but what's the point of including the mistype?  Here is my output from the current `dev` build of my cut-down version:
```
> =node.heap()
44920
> decoded=cjson.decode [[{"value" : "!b", "timestamp":1456955824}]]
> for k,v in pairs(decoded) do print (type(k),k,type(v),v) end
string  value   string  !b
string  timestamp       number  1456955824
> local new_stamp = decoded['timestamp'] print (new_stamp, decoded['timestamp'])
1456955824      1456955824
> 
```
Still falls under #719 so no reopen.  Your test case has to be repeatable, so keep it simple and supply a full test case and output

pjsg, 2016-03-03:
I also tried downloading your build, and it behaves correctly.

```
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: cjson,file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-02-26 05:26
 powered by Lua 5.1.4 on SDK 1.4.0
lua: cannot open init.lua
> last_stamp = 0
> function parseCmd(response)
>> decoded = cjson.decode(response)
>> if last_stamp < decoded["timestamp"] then
>> local new_stamp = decoded["timestamp"]
>> print(decoded["timestamp"])
>> print(new_stamp)
>> end
>> end
> =parseCmd("{ \"timestamp\" : 1456955824 }")
1456955824
1456955824
```

jaccarmac, 2016-03-03:
Hmmm... Thanks for the tests. If it's not repeatable, I'll have to look into other causes!