nodemcu/nodemcu-firmware #1112: filenames of 32 chars should either not be allowed, or handled correctly

devyte, 2016-03-03:
Try this:
```
file.open("thisisatestfilewithalongname.lua", "w")
file.write("return function(arg) return arg end")
file.close()
=dofile("thisisatestfilewithalongname.lua")(true)
cannot open thisisatestfilewithalongname.lua
```

The file is actually created on the file system, but the listing shows corrupted:
```
function ll(regex)
    if(regex == nil) then
        regex = ".*"
    end
    local regex = "^("..regex..")$"
    local file_list = file.list()
    local sorted_file_list = {}
    for n,s in pairs(file_list) do
        if(string.match(n, regex) ~= nil) then
            table.insert(sorted_file_list, n) 
        end
    end
    table.sort(sorted_file_list)
    for i,n in ipairs(sorted_file_list) do 
        print(string.format("%-40s %7s", n, file_list[n]))
    end
end

> ll("this.*")
thisisatestfilewithalongname.lua@#           35
```

The file can be opened and read:
```
function more(fn)
    if file.open(fn, "r") == nil then
        print("Can't open file "..fn)
        file.close()
        return
    end

    local line = file.readline()
    while(line) do
        if line:sub(-1) == "\n" then
            print(line:sub(1,-2))
        else
            print(line)
        end
        line = file.readline()
    end
    file.close()
end

> more("thisisatestfilewithalongname.lua")
return function(arg) return arg end
```

but not compiled:
```
> node.compile("thisisatestfilewithalongname.lua")
stdin:1: not a .lua file
stack traceback:
	[C]: in function 'compile'
	stdin:1: in main chunk

```

Renaming the file works. If the file is renamed to a shorter name, then all is ok (maybe, I guess it's possible that the file system was subtly corrupted).
This looks like an off-by-one error in C, probably related to strlen() vs. strlen()+null terminator, maybe in more than one place.

TerryE, 2016-03-03:
This condition should be thrown as an error.

TerryE, 2016-03-04:
Yup, I've had a look at this `spiffs_hydrogen.c uses `char[32]` for the name field and also uses a mix of the `strnXXX()` and `strXXX()` forms. It is possible to secure all of these to the length-safe form, but the `struct spiffs_dirent` reserves a char[32] for the name, and without an explicit length,and all of the test suites assume that the name is a standard C string -- that is null terminated -- though don't seem to check this `strlen(name) == 31` condition

My conclusion is that it would be unsafe to try to permit 32-byte long names because we would be introducing a lot of extra code paths to test.  I will therefore raise a PR which closes the issue for the firmware:

-  to add the minumum changes to SPIFFS to prevent name string overrun
-  to limit filenames in Lua to 31 chars
-  throw an error if a Lua filename contains an embedded null char.

I will leave this issue open for now because someone with a track record with the SPIFFS maintainers might want to upstream this issue.

TerryE, 2016-03-05:
Fix implemented in `dev`.

closed 2016-03-05 by TerryE