nodemcu/nodemcu-firmware #1121: wifi.sta.setip() ignores gateway parameter

eku, 2016-03-05:
I'm using HEAD from the dev branch

```NodeMCU 1.5.1 build 20160305 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)```

and configure the wifi module in station mode

```
> =wifi.sta.setip({ip="192.168.33.131",netmask="255.255.255.0",gw="192.168.33.1"})
true
> =wifi.sta.getip()
192.168.33.131  255.255.255.0   0.0.0.0
```

The ESP runs an UDP server, which replys on received packets. As long as the client is in the same sub-net it will receive the reply. In my use case the client is in a different sub-net and never gets any reply. I guess the ESP drops the reply packet because of the missing gateway.

closed 2016-03-05 by eku

marcelstoer, 2016-03-06:
@eku why did you close this without further comments? Because you found out the gateway parameter as per http://nodemcu.readthedocs.org/en/dev/en/modules/wifi/#wifistasetip is called `gateway` instead of `gw`?

eku, 2016-03-06:
Good guess!