nodemcu/nodemcu-firmware #1124: tmr.alarm not firing the callback if registered in timer callback

wuyuanyi135, 2016-03-07:
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: adc,bit,cjson,file,gpio,mqtt,net,node,ow,pwm,rtctime,sntp,tmr,uart,wifi
 build 	built on: 2016-03-06 12:05
 powered by Lua 5.1.4 on SDK 1.4.0
```

I just found that, if I register `tmr.alarm` in the callback of another `tmr.alarm`, the internal `tmr.alarm` can only fire when its interval is shorter than the outer  alarm.

For example

```
tmr.alarm(0,1000,1,function()
    print(0)
    tmr.alarm(1,1000,0,function()
        print(1)
    end)
end)
```
Output: 
```
0
0
0
0
```

Let the internal interval a bit shorter
```
tmr.alarm(0,1000,1,function()
    print(0)
    tmr.alarm(1,990,0,function()
        print(1)
    end)
end)
```
Output: (work as expected)
```
0
1
0
1
```

If the external timer is `ALARM_SINGLE`
```
tmr.alarm(0,1000,0,function()
    print(0)
    tmr.alarm(1,1000,0,function()
        print(1)
    end)
end)
```
Output: (work as expected)
```
0
1
```

pjsg, 2016-03-07:
That is what I would expect.... 

Bear in mind that when you initiate timer 1 for the second time, it cancels the first timer. What is happening is that timer 1 is being cancelled at just about the time that it would fire. 

I claim Not-A-Bug.

TerryE, 2016-03-07:
Philip, I agree. Alarm 0 will continually reset alarm 1 so it will never fire. @wuyuanyi135 , if you stop alarm 0 then alarm 1 will fire once.

closed 2016-03-07 by TerryE

wuyuanyi135, 2016-03-07:
Thanks @TerryE  and @pjsg. I will register the event before start it.