nodemcu/nodemcu-firmware #1131: Add optional callbacks to wifi.sta.connect() and wifi.sta.disconnect()

alex3kov, 2016-03-07:
I'm aware of `wifi.sta.eventMonReg()`. I'm thinking about writing several different functions in one program. Each will connect wifi at the start and disconnect at the end. They will do different things (obviously) and `wifi.sta.eventMonReg()` can only hold one function. So I would have to register/unregister `eventMonReg` each time and it wouldn't be **quite** deterministic due to asynchronicity:
```
function test1()
  wifi.sta.eventMonReg(wifi.STA_GOTIP, function()
    --do things here
    wifi.sta.eventMonReg(wifi.STA_GOTIP, "unreg")
    wifi.disconnect()
  end)
  wifi.connect()
end
```
...what if wifi manages to connect before event monitor gets registered? I realize it's unlikely, but who knows - mega-fast WiFi router with DHCP already aware of ESP's MAC, some random delay in ESP to register event monitor and voila, `eventMonReg`'s callback won't fire. I think this would look better, be more EDD-like and deterministic:
```
function test2()
  wifi.connect(function()
    --do things here
  wifi.disconnect()
  end)
end
```

TerryE, 2016-03-07:
Alex, that's the second or 3rd time you've referred to EDD.  Can you please explain for poor old farts who have been in this business for 40 years but totally missed this acronym?  Good old google doesn't help.

The second thing is that this is mainly a wrapper around the SDK API call `wifi_station_connect()` which doesn't have any callback so you have to poll `wifi_station_get_connect_status()` either at a Lua level or in a C wrapper.  But "at 20,000ft" you are quite right programming connections is a mess from an applications programmer's PoV.

Not sure at this stage what we do about it.

dnc40085, 2016-03-08:
@TerryE I had to look it up myself and I'm pretty sure EDD means Event Driven Development.

TerryE, 2016-03-08:
OK figures. Alex is right. Developing a nodeMCU Lua app is more like developing a jQuery or DB triggered app. Giving a more rigorous temporal dependency model would be better.

dnc40085, 2016-03-08:
@TerryE I don't believe it would be that complicated to add the functionality described here after the event monitor in PR #1018 is merged, no polling of `wifi_station_get_connect_status()` would be necessary.

pjsg, 2016-03-08:
This got me to thinking -- when we get the callbacks from the Espressif SDK (for these events or net events), what environment are we in? Should we be posting to the task queue to get the system into a known state? Are there any restrictions on making calls in these callbacks?

alex3kov, 2016-03-08:
@dnc40085 @TerryE Yes, I meant Event-Driven Development; sorry for being cryptic.

TerryE, 2016-03-08:
@pjsg to be honest I don't know at what priority level the net and WiFi cb's are at our even how they are positioned relative to the user high/medium/low. Writing some Lua scripts to retro-engineer this is on my to-do list unless someone gets there first. I do know that the more recent SDK programming guides have warnings on some espconn and wifi API calls about not calling further calls in this task.

This wifi_connect issue is complicated because of the interplay with autoconnect, and the lack of wifi cb's.

closed 2017-05-21 by marcelstoer

ram-prasanth-9394, 2019-11-12:
need to connect two differnet wifi networks in node mcu.
Eg. There are two wifi connections WiFi1(**Home WiFi**) and WiFi2(**Mobil Data**)
.
.
.

For Example, Im entering in my house with my mobile data ON and as if when i enter into the home i need to connect automatically to my WiFi1(Home WiFi).........
Need a first priority For WiFi1 when ever i

ram-prasanth-9394, 2019-11-12:
need to connect two differnet wifi networks in node mcu.
Eg. There are two wifi connections WiFi1(Home WiFi) and WiFi2(Mobil Data)
.
.
.

For Example, Im entering in my house with my mobile data ON and as if when i enter into the home i need to connect automatically to my WiFi1(Home WiFi).........
Need a first priority For WiFi1 when ever i enters home

HHHartmann, 2019-11-12:
Please see #1010
(Would close but as it is closed already for over 2 years ...)