nodemcu/nodemcu-firmware #1134: Unsubscribe gpio.trig

UncleRedz, 2016-03-08:
This may have been an undocumented feature, but with the old gpio implementation, you could unsubscribe the callback for a specific pin by writing something like this, 

gpio.trig(2,"none")

As the word "none" was not recognised, the code simply removed the callback. 

Running the same code on dev now after TerryE rewrite, the ESP8266 instantly reboots. 

I don't see in the docs that there is any way to unsubscribe, but it would be nice if it was possible. At the very least, the instant reboot should be fixed. 

(On a side note, after very brief testing, gpio trig seems more stable now.) 

Thanks for your efforts with the NodeMCU firmware.

TerryE, 2016-03-08:
You just set the GPIO mode to other than interrupt, say input and this clears down any associated triggers.  When you say "instantly reboots" I think you mean " throws a Lua error and because I haven't done a protected call, this triggers an unprotected call exception which the RTS handles by doing a PANIC restart." 

See #1119

UncleRedz, 2016-03-08:
I don't think it throws a lua error, there is no panic message, just the short sequence of garbage characters and a reboot. Looks the same as a node.restart(). Which is why I described it as an instant reboot. It looks like there is an error case in the c code that isn't handled when an unexpected word is used for what to trigger on.

TerryE, 2016-03-08:
OK, I will take a look today.

TerryE, 2016-03-08:
I have just put a fix in for this.  See #1136

Sorry but I done want to allow "none".  The problem is that the previous algo defaulted to interrupts disabled if the option was one that it did know about.  Hence if you mistyped "up" say as "yp", this would fail silently and disable the trigger.  The correct action is to omit the argument.

TerryE, 2016-03-08:
@UncleRedz if you don't mind we're now tracking this through PR #1136 so I will close this.  Rgds Terry

closed 2016-03-08 by TerryE

TerryE, 2016-03-09:
See also #1141.  Your comments will be appreciated.