nodemcu/nodemcu-firmware #1139: can we configure the gpio push mode ?

wuyuanyi135, 2016-03-09:
In the doc
```
gpio.mode(pin, mode [, pullup])
```
I wonder if we can specify the pull mode (push pull/ open drain) here?

marcelstoer, 2016-03-09:
Not sure I really get the question. The docs seem very clear to me. You either use pullup or leave it floating, that's it.

wuyuanyi135, 2016-03-09:
So this "pull up" actually mean push-pull? Then is it possible to activate
internal pull-up resistor

devsaurus, 2016-03-09:
@wuyuanyi135 looking at the code in `set_gpio_no_interrupt()` of platform.c I'd say that the output driver stages of the GPIOs are always configured to push-pull mode when set to output with `gpio.mode()`. There's currently no way to put them into open-drain mode.

The `pullup` parameter you mentioned controls whether the weak pull-up resistor is enabled or not.

Adding open-drain capability (and improving the docs) sounds like a valid improvement.

wuyuanyi135, 2016-03-10:
@devsaurus Thanks. Great to implement push mode configuration

pjsg, 2016-03-10:
The hardware supports all four pin settings (LO, HI, Hi-impedance, weak pullup).

TerryE, 2016-03-10:
See the GPIO V0.5 Spec ([here](http://bbs.espressif.com/viewtopic.php?f=51&t=819]) I believe a later version is also available.

devsaurus, 2016-03-10:
Thanks for the links. I'll follow up with a PR for adding open-drain mode.

devsaurus, 2016-04-10:
The respective enhancement was merged some time ago.

closed 2016-04-10 by devsaurus