nodemcu/nodemcu-firmware #1144: Upgrade SDK to 1.5.2

karrots, 2016-03-10:
PR #917 mentioned waiting for 1.5.2. It seems this is out now. Not sure I have the expertise to create a PR for it but figured it should be tracked. 

http://bbs.espressif.com/viewtopic.php?f=46&t=1702

jmattsson, 2016-03-10:
I have it on my to-do list as soon as work quiets down a bit over the next couple of weeks.

There was nothing of particularly high importance introduced/fixed in 1.5.2, so I haven't felt bad about not getting it in yet :)

pjsg, 2016-03-10:
It is fairly easy (now) to move to a new SDK. The issue with 1.5.2 is that the initialization bytes have changed, and there is a replacement libssl already.

TerryE, 2016-03-10:
Yup, it's easy if you have time to do it.  There's just some i's to be dotted and t's to be crossed on each upgrade and it ends up taking more than a few hours work.  As Johny says, there's nothing in 1.5.2 that cries out for an early upgrade, so I don't see this as an urgent priority.  What concerns me more if that `dev` has been so active over the this last couple of months that `master` is now too far behind.

jmattsson, 2016-03-10:
Yeah, we should consider doing a master drop every ~2 months at least I think. There's always going to be something in flux, but unless we want to really curate master I don't see much choice.

karrots, 2016-03-11:
Yea ok I forget it's all at the top of the Makefile now. The harder part is the patch released after that @pjsg is referencing. It's too bad a 1.5.3 isn't in the wings before the snap to master.

dnc40085, 2016-03-11:
I was looking at the programming guide [2C-ESP8266-SDK__API Guide__EN_V1.5.2_20160308](http://bbs.espressif.com/viewtopic.php?f=51&t=1024) and I found that the SDK 1.5.2 timers `os_timer_arm()` are limited to about 1.91 hours, down from the limit of about 74.57 hours that SDK 1.5.1 offers

karrots, 2016-04-18:
Closing issue as 1.5.3 was released yesterday. Some of the items in the release notes include:

- Moved `driver_lib` folder to be as `ESP8266_NONOS_SDK\driver_lib`.
- Added SPI driver, refer to `ESP8266_NONOS_SDK\driver_lib\driver\spi_interface.c`.
- Optimized espconn.
- Optimized DNS function.
- Optimized mDNS function.
- Optimized the disconnection when ESP8266 runs as TCP server.

TerryE, 2016-04-18:
It's not clear to me what "optimied" means here, and whether it reality this is a bugfix. It does mean changed though. All we can do if take a look at the open source elements and any possible API changes and make cac can on whether we take dev to 1.5.3 soon after we have done the dev snap.

TerryE, 2016-04-18:
@pjsg Philip, perhaps you could take a close look at the mDNS and DNS elements?

pjsg, 2016-04-23:
The mdns code eliminates a memory leak (which I had effectively eliminated anyway).

The dns code change again looks like fixing a memory leak in some error case. Not quite sure what. Probably worth including....

marcelstoer, 2016-06-29:
What's everyones preference here?
- close this
- keep renaming from 1.5.n to 1.5.n+1, if so it's 1.5.4 now
- freeze the SDK until NodeMCU is based on RTOS

djphoenix, 2016-06-29:
So, I prefer to keep updating through NONOS SDK until we reach working RTOS build.
Also I propose rejection of all PRs that based on espconn layer (because it deprecated in RTOS). Maybe exception for bugfixes, but not for new modules.

marcelstoer, 2016-06-29:
I agree, we just need to find someone capable and willing to do the 1.5.4 upgrade...

djphoenix, 2016-06-29:
In my local repo I updated to 1.5.3, I think 1.5.4 not too different. Maybe PR some later.

jmattsson, 2016-06-30:
I'd normally stick my hand up for it, but I've been focusing on the RTOS and ESP32 side (what little time I currently have available).

devyte, 2016-06-30:
I'd also prefer to continue with the NONOS SDK, at least for now. While I'd love to see the RTOS working and check out what it can do, I suspect it will reduce the available ram even further, and I'm already struggling with that.

TerryE, 2016-07-06:
Let's carry this on in #1377.  There's no point in having two topics open on the same agenda,

PR #1377 (2016-07-10): Update to SDK 1.5.4
Build success, bootup success, seems like all features keeps working.
Seems like CI using "esp-open-sdk.zip" from tools folder, maybe needed to update also. 

Fixes #1144, #884.
closed 2016-07-06 by TerryE