nodemcu/nodemcu-firmware #1146: Propagate `dev` to `master`

marcelstoer, 2016-03-10:
We should IMO plan ahead and come up with a preliminary list of issue still to be fixed before current `dev` can be propagated to `master`.

Here's my take. Fellow committers feel free to edit this list directly here. Big question for me is if we should wait for SDK 1.5.2 support or not.

- [ ] PR #1116, Fix crash bug when HTTP response from server is NULL.
- [ ] PR #1123, MQTT Client - CONNACK processing
- [ ] PR #1129, Try to make qos~=0 work better
- [ ] PR #1136, one line bugfix to GPIO.c -- Oops!
- [ ] PR #1138, Improve overall enduser_setup user experience. 
- [ ] PR #1142, Fix null-pointer crash on early input
- [ ] PR #1143, fix unexpected low level after writing
- [ ] bug #1065, wifi.sta.eventMonReg previous_State is always "nil"
- [ ] bug #1039, rtctime.dsleep doesn't set correct bootreason
- [ ] #1009, Thoroughly document how NodeMCU handles esp_init_data_defaults
- [ ] #966, How to handle backward compatibility in WS2812 module

TerryE, 2016-03-10:
Marcel, thanks for this analysis. I found it very useful.  Whilst I broadly agree with the list, ultimately I also feel that we should apply some sort of 90/10 or 80/20 rule here, otherwise will never get to then state where `master` is stable enough.

marcelstoer, 2016-03-11:
Terry, I don't think you'll find a fellow committer who disagrees on applying Pareto principle here :wink: However, I'm a bit puzzled...What changes exactly do you propose? Which issues do you think are missing, which ones would you remove?

TerryE, 2016-03-11:
Marcel, it's just that since we as a group of supporters ave started to actively work on the firmware -- that is really over this last 12 months, the churn of active development has really stepped up a couple of notches.  You list off 13 PRs that there are good arguments for clearing before we snap `dev` to `master` but by the time they are cleared, then there will be another 12 for which the same argument applies and so on, so there will never be a clean cut, unless we tweak the process, IMO.

-  We either just do a reasonable cut at some point and accept that it won't be "perfect", but instead a good step-wise improvement n the previous cut, or

-  We have a conscious "about to cut to dev" process where we do what you suggest an identify a cut of outstanding PRs and only promote them and perhaps one or two other "sound" ones, but either way this will involve either 
   -  either creating an extra temporary branch for a couple of weeks, where we cut dev -> pre-master, and then selectively merge in agreed commits into this, whilst dev itself continues on its more free running commit process, or
    -  we say to everyone "for the next (two) weeks" our priority is to close and merge PRs X, Y, Z and all others will be held.

I am not strongly advocating any one of these.  I think that this should be a consensus approach.  It just the release management -- however we approach it -- has wrinkles that we can't avoid.

marcelstoer, 2016-03-11:
> there will never be a clean cut, unless we tweak the process, IMO.

Indeed, that was the reason I created this issue. We can't just keep pushing to `dev` without end and not think about how/when to propagate to `master`. I feel that *any* plan as for how to do this is better than no plan at all.

I favor your 2b approach (identify a cut of outstanding PRs and hold back others).

>  the release management -- however we approach it -- has wrinkles that we can't avoid

That's fine, I'm willing to accept that and make the best out of it.

TerryE, 2016-03-11:
So we're going o be saying: sorry for the delay -- see \#1146, for the next few weeks eh?  :-)

I'll go through this list and clear off any that I am holding up.

devsaurus, 2016-03-12:
Marcel, your list of issues that should be resolved before an update of `master` is very helpful, thanks. It's important to define a goal for content freeze on the stable branch IMO.

I'd favour
> creating an extra temporary branch for a couple of weeks, where we cut dev -> pre-master

but have no clue how this is supported by git to avoid another merge mess between `master` and `dev` when `dev` is progressing concurrently to settling of `pre-master`. Is there anyone around with enough git know-how to consult here?
If there's no clear workflow, then let's fall back to the 2b approach.

TerryE, 2016-03-12:
AFAIK, you just clone dev at the start of the cut to `pre-master` then let `dev` continue as before.  PRs are selectively cherry-picked to `pre-master` and when we are ready to go we just delete the old `master` and rename `pre-master` to `master`.  Anyway, I am not a git expert, but that's how I'd do it.

dnc40085, 2016-03-13:
@marcelstoer PR #1154 should allow you to check at least one item off your list

pjsg, 2016-03-22:
I'd just create an integration branch off dev (called integration or pre-master). Can selectively pull stuff onto that branch. At the appropriate instance, we create the PR from integration to master and someone presses the merge button and we drink champagne. 

I hate renaming branches as it tends to screw up clones of the repo.

marcelstoer, 2016-03-22:
> when we are ready to go we just delete the old `master` and rename `pre-master` to `master``

That would probably work but would cause head ache for all those with forks of this repo.
If we indeed go for 2a and an integration branch it means a bit of extra work because one of us has to manually merge every PR we allow into `dev` to the integration branch. Also, we (or someone else) would have to test that branch explicitly. With 2b `dev` would get tested "automatically" by those 20% who choose that branch in my build service. They might of course pick the integration branch just as well but we'd have to promote its use.

TerryE, 2016-03-22:
I need to check, but AFAIK we haven't done any commits to `master` since the last clone from `dev` so merging `pre-master` to master should just be a fast-forward.

jmattsson, 2016-03-30:
I took the liberty of adding the bugfix PRs I raised today, and crossed off the now-merged EUS PR.

I'm still wondering whether 1009 would be better resolved by having a Lua-land `adc.rewrite_init107(adc.READVDD33 | adc.READADC)` function, and ditch the entire embedded ESP init data block (since the SDK now has its own copy anyway).

marcelstoer, 2016-03-30:
> I took the liberty of adding the bugfix PRs I raised today, and crossed off the now-merged EUS PR.

Thanks Johny. I nearly closed this issue yesterday because of disagreement and inactivity. As long as we haven't agreed on a procedure it does not make much sense maintaining the task list I think.

jmattsson, 2016-03-30:
Sorry, I've been crazy busy with work; only now starting to catch a wee bit on this project.

pjsg, 2016-03-30:
I'd like to get this one closed out -- or rather, I'd like to get dev merged to master. Yes, it isn't perfect, but it is (IMHO) better than the current master.

TerryE, 2016-03-30:
On balance, I think that Philip is correct. None of the remaining PRs are essential. This current cut would be a far better master.

marcelstoer, 2016-03-30:
> Yes, it isn't perfect, but it is (IMHO) better than the current master.

True, but that same argument is valid after the merge of nearly every PR :wink:.

To me it'd really make sense to wait for the remaining 3 PRs plus #1157 which is not on the list yet. Both the EUS and the HTTP module are becoming more and more popular and I'd like to accommodate for that by waiting for the PRs. Assuming we have support from @jfollas and @vowstar this shouldn't take weeks.

jmattsson, 2016-03-31:
Is anyone across the OW stuff enough to help out on 1143? It seems a trivial fix (plus @pjsg 's requested doc update), but I'm not feeling qualified in this area.

sekarpdkt, 2016-04-10:
Would you also move encoder to master? I think encoder is not yet in master. I may be wrong as I am using online firmware builder.

marcelstoer, 2016-04-10:
@sekarpdkt you get this automatically once the current `dev` is merged to `master`. In fact, you get *everything* that's been committed to `dev` since Dec 28th, 2015.

pjsg, 2016-04-24:
I'd like to include the coap fix: #1260 as it prevents a stack overflow....

jmattsson, 2016-05-11:
Just a note here about PR #1285 in relation to 1009. Please review and feedback.

marcelstoer, 2016-05-11:
Johny I was wondering when I saw #1285 whether you wanted to see that merged (instead of #1009) before the next dev-to-master snap? The comment above seems to suggest that, right?

jmattsson, 2016-05-11:
I think it's the best compromise giving the variables we have to work with. It means the user has to know less about init data and can easily copy the example from the docs and end up with the ADC in the desired mode, consistently. Whether right before the drop to master is the right time I'm not exactly 100% on - I'd like to have at least a couple of other people give this a good testing and see if they can break it. My testing suggests it's doing its job, but "it works on my machine" isn't quite sufficient here I reckon ;)

marcelstoer, 2016-05-11:
I fully agree but I'm currently not in a position to test myself (on holidays). Current status:

- I removed #1143 from the above list because @karrots, @pstolarz and @wuyuanyi135 are discussing more fundamental issues with the 1-wire module now. 
- I also replaced #1009 with #1285. 
- Even though #1157 would be really nice to have I understand @jfollas doesn't have enough time right now to give it another look. Hence, no use adding it to the list.

All in all this means that once #1285 has landed we'd finally be ready to merge.

jmattsson, 2016-05-11:
Apologies for holding things up for so long!

marcelstoer, 2016-05-11:
Why? It's not your fault. That's just how it is with OSS and a bunch of volunteers.

pjsg, 2016-05-11:
I'd like to get the mDns change #1239 merged in as well. It cleans up a number of issues.

karrots, 2016-05-11:
@marcelsoer I think we should get a basic patch integrated into master the current OW code, is broken unless using parasitic mode. The fundamental change I brought up would likely break most of the DS18B20 Arduino type demo's out there and would probably be best propagated upstream as well so it's a longer discussion.

I'll remove the noop's from the code then create a PR for others to comment on. Not quite sure the best route to get the documentation proposed in #1143 integrated.

devsaurus, 2016-05-23:
Now that #1285 was merged, is there anything left which is considered as gating for the snap?

marcelstoer, 2016-05-23:
Nothing except for PR #1291 which is a partial fix for #1143, @nickandrew wanted to verify the PR a few days ago.

PR #1336 (2016-06-03): 1.5.1 master drop
While the big guys are busy with ESP32 and RTOS SDK I want to finish up what I started three months ago: this fixes #1146.
Eventually #1291 didn't make the cut. It just doesn't seem justified waiting any longer and hoping for feedback.
closed 2016-06-03 by devsaurus