nodemcu/nodemcu-firmware #1149: Spiffs cannot hold large file (larger than around 26857B)

MarsTechHAN, 2016-03-11:
Happened on both master branch and dev branch.
MASTER Commit: c8037568571edb5c568c2f8231e4f8ce0683b883

 If I use old firmware(such as 0.9.6 release) it can write all but return wrong length.

I want to write the GBK.bin(A dot matrix font lib for u8g to display Chinese word) into the Flash through SPIFFS. 

I try the ESPlorer's upload, but the nodemcu will stop response after around 26857B ( at a static point).
`Uploading to ESP file gbk.bin...Waiting answer from ESP - Timeout reached. Command aborted.`

Then I used mkspiffs, but when I use esptool to write the output bin inside, the nodemcu cannot detect the file, the filename is not ascii charater and size is -678239427394(a large minus number) 
but I can open it and read/write. （Have not check weather the data of the file is correct or not yet)

`Uploading to ESP file gbk.bin.001...Success
Uploading to ESP file gbk.bin.002...Success
Uploading to ESP file gbk.bin.003...Success
Uploading to ESP file gbk.bin.004...Success
Uploading to ESP file gbk.bin.005...Success
Uploading to ESP file gbk.bin.006...Success
Uploading to ESP file gbk.bin.007...Success
Uploading to ESP file gbk.bin.008...Success
Uploading to ESP file gbk.bin.009...Success
Uploading to ESP file gbk.bin.010...Success
Uploading to ESP file gbk.bin.011...Success
Uploading to ESP file gbk.bin.012...Success
Uploading to ESP file gbk.bin.013...Success
`
Now I split it into 42 files, each 24KB, and looks good :X but it is really inconvience, wish to solve the problem~ :D

devsaurus, 2016-03-11:
This seems to be related to #898. Though there's no fix at the moment.

MarsTechHAN, 2016-03-11:
Thanks for the reminding,  In addition, with the circumstance 2, I can open the file, read it but it does not show correct on the file.list() with looks like overflood number, which is really weird?

pjsg, 2016-03-17:
I am looking to replace the current version of spiffs with the current master version from the source repo. This can handle large files....

marcelstoer, 2016-06-05:
Fixed by #1226.

closed 2016-06-05 by marcelstoer