nodemcu/nodemcu-firmware #1152: HTTP module - binary download issue

jfollas, 2016-03-11:
Download binary files using the HTTP client is failing for me when there's 0x00 in the data. Pretty sure it's because the buffer is a char* in the HTTP client and C null termination is coming into play.

Example: httpclient.c
``` c
static void ICACHE_FLASH_ATTR http_receive_callback( void * arg, char * buf, unsigned short len )
{
	struct espconn	* conn	= (struct espconn *) arg;
	request_args_t	* req	= (request_args_t *) conn->reverse;

	if ( req->buffer == NULL )
	{
		return;
	}

	/* Let's do the equivalent of a realloc(). */
	const int	new_size = req->buffer_size + len;
	char		* new_buffer;
	if ( new_size > BUFFER_SIZE_MAX || NULL == (new_buffer = (char *) os_malloc( new_size ) ) )
	{
		HTTPCLIENT_DEBUG( "Response too long (%d)\n", new_size );
		req->buffer[0] = '\0';                                                                  /* Discard the buffer to avoid using an incomplete response. */
		if ( req->secure )
			espconn_secure_disconnect( conn );
		else
			espconn_disconnect( conn );
		return;                                                                                 /* The disconnect callback will be called. */
	}

	os_memcpy( new_buffer, req->buffer, req->buffer_size );
	os_memcpy( new_buffer + req->buffer_size - 1 /*overwrite the null character*/, buf, len );      /* Append new data. */
	new_buffer[new_size - 1] = '\0';                                                                /* Make sure there is an end of string. */

	os_free( req->buffer );
	req->buffer		= new_buffer;
	req->buffer_size	= new_size;
}
```
Would the fix be to just make this uint8_t * instead (and also change everywhere that the buffer is passed through the code)?

jfollas, 2016-03-12:
I have a solution for this that I'll prepare a PR for.

TerryE, 2016-03-12:
`lua_pushstring()` should _never_ be used for user data, for this reason.  It should be limited to pusihing  known patterns and text which can't contain an embedded nullchar.  Having done a quick scan of `app/modules` this is the only case here.

jfollas, 2016-03-14:
Little hiccup in the process of getting here, but the PR 1157 includes the fix for this issue.

TerryE, 2016-03-14:
@jfollas  @vowstar, I've just had a look at this source and it has a lot of short comings, for example:
-  it requires you to build with SSL enabled even if you are only using HTTP in your app.
-  it reimplements standard functions for no clear reason
-  it's memory allocation / freeing is flaky so it will generate memory leaks
-  it doesn't handle records which span IP packets.

Nothing to do with this PR but a lot more work is really needed on this.

jfollas, 2016-03-14:
@TerryE: Having spent time in the code to troubleshoot/fix the issue, I'd agree that the http module could really use more effort. My PR focused on the tasks at hand (i.e., all of the things that you mentioned exist in the current module's code even without the corrective parts of my PR).

Oh, and I think it does handle the scenario where the payload arrives in different packets (`http_receive_callback`). The finalization work isn't until the socket disconnect callback.

closed 2017-02-18 by marcelstoer