nodemcu/nodemcu-firmware #1159: Corrupt SDK downloaded from Espressif breaks build

marcelstoer, 2016-03-14:
Starting from this morning all `dev` builds in my build service fail.

> sha1sum: WARNING: 1 computed checksum did NOT match

Turns out the file we download from http://bbs.espressif.com/download/file.php?id=1046 does not just have an unexpected SHA1 checksum, it isn't even a valid ZIP file.

> /tmp > unzip -t esp_iot_sdk_v1.5.1_16_01_08.zip 
Archive:  esp_iot_sdk_v1.5.1_16_01_08.zip
warning [esp_iot_sdk_v1.5.1_16_01_08.zip]:  7637 extra bytes at beginning or within zipfile
  (attempting to process anyway)
error [esp_iot_sdk_v1.5.1_16_01_08.zip]:  start of central directory not found;
  zipfile corrupt.
  (please check that you have transferred or created the zipfile in the
  appropriate BINARY mode and that you have compiled UnZip properly)

-> all CI builds Travis runs for submitted PRs will fail as well

TerryE, 2016-03-14:
There is something ***wrong*** with the archive.  The `nodemcu-firmware/cache/esp_iot_sdk_v1.5.1_16_01_08.zip` on my VM is 2,185,752 bytes long as opposed to the new version with this URI which is 2,193,389 bytes (SHA1 374f689a5f9e47690d7b4cd2fc1a1094f3fd5a4f) -- that is an extra 7,637 as unzip detects.  I suspect that the site has been hit and this now has a virus payload design to attack M$ Windows.  Worth logging a warning on the Espressif BBS.

marcelstoer, 2016-03-14:
I came to the same conclusion. I posted to Espressif BBS right after I created this issue but since a moderator needs to release my post it's gonna take a while (it's already dark in China). To raise awareness I also sent an email to my contact at Espressif.

dnc40085, 2016-03-14:
Must have been a hack, all the pictures I can find in the forums are corrupt, Like the ones [here](http://bbs.espressif.com/viewtopic.php?f=57&t=1440).

jmattsson, 2016-03-15:
I guess we've now encountered the big downside to *not* having our own copy in the repo :(

Should we go back to the way we had it before?

jmattsson, 2016-03-15:
Actually, this looks like a webserver misconfiguration. Comparing pre & post files in a hex editor I see that 0x0A has turned into 0x0D 0x0A, which is LF -> CRLF translation. It's the ol' bugbear of raw vs cooked mode again.
...
Alright, I've confirmed it. Running `dos2unix -f esp_iot_sdk_v1.5.1_16_01_08.zip` yields a working zip file, identical to the pre version. We possibly could add something like `unzip -t $file || dos2unix -f $file` to the Makefile, but that of course relies on `dos2unix` being installed. Really Espressif should just stop fscking up their web config.

TerryE, 2016-03-15:
My vote would be to keep these blogs, the SDKs and the ESP toolchain in a separate repo.  This keeps the main repo from bloating up but avoids having to download blobs routinely through the Chinese firewall. This is a performance issue only.

MarsTechHAN, 2016-03-15:
I've contact the espressif side, they said they were switching the BBS to new servers, some unexpected problem happened but unawared yesterday. Now they are haddling this problem, and will deal with it today. (All time appeared here are UTC+8，which means they will fix it at 12:00 UTC 2016/3/15 if nothing worse is found)

marcelstoer, 2016-03-15:
All good again AFAICS.

closed 2016-03-15 by marcelstoer

NicolSpies, 2016-03-15:
A NodeMCU float custom build from dev after the fix does not start up (just garbage). Build repeated with same results.

marcelstoer, 2016-03-15:
I guess that's because you haven't switched to 115'200bps, have you? See #1127 and http://nodemcu.readthedocs.org/en/dev/en/upload/.

NicolSpies, 2016-03-15:
Aghh, I have tried all the baud rates but stopped before 155200, will quickly re-flash and try again. Thanks

NicolSpies, 2016-03-15:
Thank you sir, all good !