nodemcu/nodemcu-firmware #1160: GPIO interrupt slow

maxhack, 2016-03-15:
Am using Arduino IDE to write programs for nodemcu, latest application was supposed to be an AC light dimmer using zero detect circuit, AC frequency is 50 Hz thus zero cross is detected every 10 mS (2 crosses in each cycle)

light is dimmed correctly but keeps flickering every few seconds, when I checked with oscilloscope I found out that interrupt does not occur after same exact time after zero cross detection, it has a noticeable error of around 120 uS (I also used function micros() to check each time interrupt occurs, it gave same result as digital oscilloscope)

I think that this instability causes light flickering, any ideas how to solve?

I send commands to light via normal webserver where ESP reads URL and extracts needed dimming level from it, the interrupt code I use is written below:

void zero_crosss_int(){
dimtime = (75*light_val); // For 60Hz =>65 
delayMicroseconds(dimtime); // Wait till firing the TRIAC 
digitalWrite(L1, HIGH); // Fire the TRIAC
delayMicroseconds(10); // triac On propogation delay 
// (for 60Hz use 8.33) Some Triacs need a longer period
digitalWrite(L1, LOW); // No longer trigger the TRIAC (the next zero crossing will swith it off) TRIAC
}

In setup function I use this code:

pinMode(zero_pin, INPUT_PULLUP);
attachInterrupt(zero_pin, zero_crosss_int, FALLING);

Thanks in advance.

marcelstoer, 2016-03-15:
> Am using Arduino IDE to write programs for nodemcu

Well, this is the NodeMCU *firmware* project (Lua). You need to seek help elsewhere.

closed 2016-03-15 by marcelstoer

Ondreyk, 2016-03-15:
You need use PWM. 
http://nodemcu.readthedocs.org/en/dev/en/modules/pwm/