nodemcu/nodemcu-firmware #1161: SPI - CPHA is inverted when CPOL=1

jfollas, 2016-03-15:
NodeMCU received CPOL=1 support a few months ago (before then, we were stuck with only CPOL=0, or low-idle clock). Today I scoped out one of our SPI interfaces that was misbehaving while using CPOL=1,CPHA=1, and it was showing the wrong waveform. The CPOL=1 waveform looked like what CPOL=0 should, and vice versa (it was one of those "How did this ever work in the first place?" situations).

CPOL=1, CHPA=1
[image]

CPOL=1, CPHA=0
[image]

Example of what it's supposed to look like:

[image]

I'm going to make the correction that I think, and then scope out the new waveforms. If they are correct, then I'll submit the PR.

TerryE, 2016-03-15:
@jfollas much appreciated.  I await your analysis with interest.  Regards Terry

jfollas, 2016-03-15:
Submitting a PR on this. I've verified the waveforms for all four modes.

*Mode 3*
[image]

*Mode 2*
[image]

*Mode 1*
[image]

*Mode 0*
[image]

jfollas, 2016-04-03:
#1162 has been merged.

closed 2016-04-03 by jfollas