nodemcu/nodemcu-firmware #1163: http.get() random failure

niston, 2016-03-16:
I'm periodically calling http.get(someURL) from a timer. After some random amount of time/calls, it the call will fail with code -1. If that happens, checking "someURL" from the same network as the ESP sits on, reveals no problem at all; the URL isn't unavailable or some such. As more time passes and more calls to http.get() are issued, the error suddenly goes away, comes back, goes away etc.

marcelstoer, 2016-03-16:
If you suggest this is a firmware bug (i.e. HTTP module bug) then we expect a small reproducible test case.

niston, 2016-03-22:
This is the small test scenario:

```
tmr.alarm(1, 60000, 1, function()
  http.get("http://some.server.com/textfile.txt", nil, function(code, data)
    if (code == 200 or code == 302) then
      print("OK: " .. data);
    else
      print("FAIL (Code " .. code .. ")");
    end
  end)
end)
```

I've further checked that the ESP8266 remains available via WiFi (ping to ESP IP) when this problem occurs, so it is definitely not due to a failing/bad WiFi connection.

On a side note: Would it be possible to expose the timeout value as a parameter in the function call?

marcelstoer, 2016-03-22:
Is it relevant what "http://some.server.com/textfile.txt" is?

niston, 2016-03-22:
Apparently not. In my case, it is a short string "ID=SIGNLR;R=0;Y=0;G=1;S=0" with no CRLF at the end. I noticed a weird issue with superfluous leading characters when this content (.txt file) came from an Azure hosted IIS. But as I switched to an .aspx page to serve the string, it went away. So probably just a fluke.

marcelstoer, 2016-03-22:
> In my case, it is a short string "ID=SIGNLR;R=0;Y=0;G=1;S=0" ...

I wasn't talking about the *content* of `http://some.server.com/textfile.txt`, sorry.

> Could I private message you somewhere (here, IRC) to give you my exact test URL?

Wouldn't make much sense as you probably want more than a single person trying to figure out whether this is a bug in the firmware or not. That's why I said initially that we need a reproducible test case and that includes testing against a public resource we all have access to. How about http://nodemcu.com/favicon.png?

niston, 2016-03-22:
Ok, here's the URLs Im using for testing:

https://vesmic-webshop.azurewebsites.net/signaleer.aspx (ASPX page on Azure)
http://vesmic-webshop.azurewebsites.net/test.txt (TXT file on Azure)
http://users.elis.ugent.be/~bgplovie/niston.txt (TXT file on some friend's webserver)

Something else I noted is that when http.get() test.txt from the Azure link, it has superfluous leading characters in the content: "ï»¿ID=SIGNLR;R=0;Y=0;G=1;S=0". Those superfluous chars are not there when checking with browser or wget. Also, these chars do not appear when http.get() the signaleer.aspx page instead.

marcelstoer, 2016-03-22:
Your test case run in isolation does not produce any errors. I lowered the interval to 30s and ran it for 15min. I also print the current heap size with every output to show the absence of a memory leak.

```lua
tmr.alarm(1, 30000, 1, function()
  http.get("https://vesmic-webshop.azurewebsites.net/signaleer.aspx", nil, function(code, data)
    if (code == 200 or code == 302) then
      print(tmr.now().."("..node.heap()..") : OK: " .. data);
    else
      print(tmr.now().."("..node.heap()..") : FAIL (Code " .. code .. ")");
    end
  end)
end)
```

```
 30595122(43896) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
 60599689(43856) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
 90670213(43856) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
120612016(43848) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
150607412(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
180594868(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
210599670(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
240603747(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
270607256(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
300596217(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
330604405(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
360596335(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
390594744(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
420593131(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
450596271(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
480613997(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
510604511(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
540598555(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
570597732(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
600612602(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
630602371(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
660588042(43888) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
690604495(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
720600762(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
750621005(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
780597158(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
810626838(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
840608587(43872) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
870600888(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
900592699(43880) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
```
This was run on a NodeMCU devkit V2 without any peripherals using a recent build from the `dev` branch.

And as for your statements about "superfluous characters" you should check the files with a hex editor. The three files you listed are all different, the size is between 25bytes and 28bytes (downloaded with wget on OS X).
[image]

closed 2016-03-22 by marcelstoer

niston, 2016-03-22:
This is what the simple test case looks for me:

> dofile("test.lua")
> 40034041(44056) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
78820715(44416) : FAIL (Code -1)
99100699(43936) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
129081753(43928) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
159101899(43952) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
189088369(43912) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
219099866(43944) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
249080635(43944) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
279577073(43960) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
309095163(43952) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
348823239(44432) : FAIL (Code -1)
369091559(43944) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
399099571(43944) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
438825052(44432) : FAIL (Code -1)
459096626(43952) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
489106399(43952) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
528822217(44432) : FAIL (Code -1)
549094342(43952) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
579091629(43944) : OK: ID=SIGNLR;R=0;Y=0;G=1;S=0
615227429(44864) : FAIL (Code -1)
645275501(44712) : FAIL (Code -1)
675273593(44600) : FAIL (Code -1)
705271732(44480) : FAIL (Code -1)
735269745(44328) : FAIL (Code -1)
765267821(44216) : FAIL (Code -1)
795340916(44104) : FAIL (Code -1)

pjsg, 2016-03-22:
Can you get a packet trace of what is happening? This would allow us to determine if this is a client or a server or a network problem.....

niston, 2016-03-23:
Ok, now this is esoteric: I upped the TX power on my WLAN router, and the issue appears to be gone. At least, the simple test has been running for a couple minutes without fail and so does my LUA app ever since I started it some 20 minutes ago. This apparent solution strikes me as quite odd because A) a continuous ICMP Ping to the ESP revealed no unexpected packet loss and did infact even get a reply while http.get() failed and B) pushing the RST button on the nodeMCU board immediately cured the problem every time. Oh my, what do I know?

niston, 2016-03-23:
I dunno what to say. If I put the following function into my code, notably *without* ever calling that function from anywhere, nodeMCU restarts on -you guessed it- http.get(). But -you didn't guess this- always only after the second invocation.

```
function CheckWiFiRSSI()
--get RSSI for currently configured AP
    ssid, tmp, bssid_set, bssid=wifi.sta.getconfig() 
    scan_cfg = {}
    scan_cfg.ssid = ssid 
    if bssid_set == 1 then scan_cfg.bssid = bssid else scan_cfg.bssid = nil end
    scan_cfg.channel = wifi.getchannel()
    scan_cfg.show_hidden = 0
    ssid, tmp, bssid_set, bssid=nil, nil, nil, nil
    wifi.sta.getap(scan_cfg, 1, function(t)
        for bssid,v in pairs(t) do
            local ssid, rssi, authmode, channel = string.match(v, "([^,]+),([^,]+),([^,]+),([^,]*)")
            print("CURRENT RSSI IS: "..rssi)
        end
    end)
end
```

Simply removing the function from the LUA code, the restart does not happen anymore.

pjsg, 2016-03-23:
This is beginning to feel like a memory allocation issue. Adding that code will increase the size of some memory allocation and move stuff around. It might be the case that one of the allocation calls now fails. Another possibility is that there is a memory overwrite somewhere in http.get and now it is hitting something more important.

What does node.bootreason() return after one of these restarts? The third parameter is the program counter (in decimal). If you convert to hex, it can then be looked up in the app/mapfile.... Maybe the function will give a clue.

niston, 2016-03-23:
It has been fixed by formatting the filesystem and reuploading the init.lua, as suggested by someone on IRC.

devyte, 2016-03-23:
@niston you said that when you use the asp server it seems to work. @marcelstoer apparently tested with that server for an extended period and it also worked. @marcelstoer also said that the other two servers send a different string, which he verified by downloading the txt file with wget from his pc and looking at the result with a hexeditor. Did you verify this? If the string from the other servers is messed up...
Also, per @pjsg, please execute node.bootreason() after the restart, convert the 3rd number to hex, and look that up in app/mapfile to know where the crash is occuring.

niston, 2016-03-23:
@devyte: Yeah the fluke (extra chars) on the Azure served .txt File was indeed caused by the server. I have no idea why this is, the test.txt I uploaded to Azure does not contain these extra chars. So, this was definately not a problem of http.get(). 

Also, after @marcelstoer showed that http.get() was indeed working as expected, I looked elsewhere for the problem that was causing my http.get() requests to fail. The issue was resolved by simply cranking up the router's TX power. Consequently, I was trying to implement a way to determine signal strenght from the ESP so as to recognize this sort of problem in the future (as described earlier, ICMP Ping requests **to** the ESP succeeded all the while http.get() requests **from** the ESP failed, apparently due to signal strenght).

When I did this and added the function posted above, I noticed that the ESP would restart everytime http.get() was called a second time - even without me calling said function from anywhere. Removing said function from the LUA code fixed the restarts. Someone on IRC suggested to cure this voodoo behaviour by formatting the ESP filesystem and reupping my code (including said function). I did, and this appeared to have worked at first, but then the unintended restarts came back. So, I removed said function from the LUA code again, and my application works happily ever since. I will try to reproduce this later in the evening and post my results per @pjsg 's suggestion, including what node.bootreason() says.

pjsg, 2016-03-23:
As a side note, the new version of the file system that I intend to commit after the current dev->master merge will (largely) take care of corrupt file systems. It will format the file system if it doesn't look valid....