nodemcu/nodemcu-firmware #1164: Upgrade to current version of SPIFFS

pjsg, 2016-03-17:
We have various issues with spiffs at the moment. Large files, crashes etc. Our version has diverged from the original github repo. I propose to bring our version up to date *and* make it easy to pull new versions in the future.

Also, after discussions with the SPIFFS author, I'm going to change the parameters for the default spiffs configuration. This will be somewhat more efficient than the current settings. 

In order to make spiffsimg work nicely, I want to bring that into the tree and build it from the same source that we use -- in this way, we are guaranteed to build compatible file systems. I also want to move to using the magic header on the file system so that there is less danger of trashing a file system if it doesn't match the size of the remaining flash. 

This will certainly take some time to settle down, but it shouldn't be *too* painful.

Thoughts?

TerryE, 2016-03-17:
In principle this sounds like a good idea, but we should do a careful review of our historic changes just in case any are valid for regressing into the new SPIFFS version.

Also my personal preference would be to move the SPIFFS test suite out of the core app tree into say `tests/spiffs/` and exclude it from the standard make altogether as this suite won't even run on the ESP hardware without some further work.

pjsg, 2016-03-17:
I was also proposing to get rid of the test suite as it didn't work anyway (and it you could make it build, it would not run on the ESP and would have different parameters).

I agree on looking at the historic changes. The good news is that the SPIFFS maintainer uses spiffs on an ESP!

pjsg, 2016-03-17:
From looking at the historic changes, the only one that still seems relevant is the one which fixes a bunch of issues with long filenames. I filed a ticket pellepi/spiffs#74 to get appropriate changes made. 

All the other changes were changes to the data structures to do with alignment. I want to leave all of those out and just use the base code (and make sure that any fixes we need are done through the base project).

pjsg, 2016-03-17:
This will resolve #896 #1147 #1149 #1150

jmattsson, 2016-03-18:
I believe the only change we'd want to be keeping is the tuning of the working buffers in order to keep RAM usage in control. The only alignment changes we're currently running with (to my knowledge) are in relation to willy-nilly casting into the internal working buffer, which hopefully would be fixed properly upstream. Most of our issues with alignment-related crashes were tracked down to be an inconsistent behaviour of the platform flash read/write routines which were fixed, and removed the need for all(?) the previous alignment hacks applied to SPIFFS.

If there are layout changes to the actual file system, please ping me when this goes in so I can update `spiffsimg` as well.

Good to hear Pelle has had time to fire up an ESP - last exchange he hadn't had a chance to play with one yet.

pjsg, 2016-03-31:
@jmattsson I included your spiffsimg into the nodemcu tree -- just so that it is kept in sync. You will probably want to merge it back into your repo. As far as I can tell, there is no guarantee of binary compatibility between different versions of the spiffs code..... This is why I wanted to bind it together. This could be changed....

My current branch is [pjsg/nodemcu-firmware:spiffsimg](/pjsg/nodemcu-firmware/tree/spiffsimg) -- this includes the new spiffs code, and also the code to build any disk images (if there is stuff in local/fs).

It is a bit of a scary sized PR, but it does remove a whole bunch of code.

xmurobi, 2016-03-31:
@pjsg I integrated your spiffsimg branch to nodemcu-firmware 1.4.0. But file.write always failed and cause reboot. Below is the debug info from spiffs which I copy from minicom :

free_obj_id: COMP min:0001 max:1ad7 compact:26
free_obj_id: COMP select index:1 min_count:0 min:0001 max:1ad7 compact:26
create: found free page @ 0001 bix:0 entry:0
open: fd 1 is obj id 801b
append: 14 bytes @ offs 0 of size -1
append: 801b load objixhdr page 0001:0000
append: 801b store new data page, 0002:0000 offset:0, len 14, written 0
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4026124a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3ffed74d, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

What problem it should be?

xmurobi, 2016-03-31:
@pjsg I also build your branch and burn it into esp8266 and the problem still exist. We can't save the code to esp through  ESPlorer.

pjsg, 2016-03-31:
@xmurobi What happens if you just build my branch as is, without trying to make it work on 1.4.0? I've been testing this on 1.5.1.

jmattsson, 2016-04-01:
> But the problem is, each time I burn the bin files to esp8266 will trigger the flash format operation at first boot.

This suggests you are not specifying the correct flash size / mode / speed when you're flashing the NodeMCU image. If you're using esptool.py, make sure you use the correct -fs -fm and -ff parameters to write_flash.

pjsg, 2016-04-01:
Actually, there was a bug in my code that I introduced with a late shuffle of the code. I have now pushed a better version. Sorry about that @xmurobi 

This is exactly why I want to get several eyes on it.....

xmurobi, 2016-04-01:
@jmattsson I do use the parameters in Makefile:
`$(ESPTOOL) --port $(ESPPORT) write_flash -fm dio -ff 80m 0x00000 $(FIRMWAREDIR)0x00000.bin 0x10000 $(FIRMWAREDIR)0x10000.bin`

jmattsson, 2016-04-01:
If you're on a NodeMCU dev-board or other ESP-12(E) module, you'll want to also add `-fs 32m` (32megabit => 4megabyte)

pjsg, 2016-04-03:
@marcelstoer There is one issue with auto-building the file system image -- I need to have gcc on the build system. Currently, your docker image doesn't include gcc, but I copied your dockerfile and added `gcc` to the list of packages to install, and then everything worked fine.

marcelstoer, 2016-06-05:
Fixed by #1226.

commit af39a0bc (2016-06-05): Upgrade to latest SPIFFS and add building of file systems (#1226)

Fixes #1164 and thus also #1150, #1149, #1147 and #898.

* Move to latest version of SPIFFS
* Add SPIFFS porting layer for NodeMCU
* Add option to delete output if it doesn't fit
* Change FLASHSIZE to be in bits by default: default 4mb 32mb
* Add SPIFFS_MAX_FILESYSTEM_SIZE override
* Add notes on SPIFFS_FIXED_LOCATION
* Add 1M boundary
* Include the current version of the LICENSE
closed 2016-06-05 by marcelstoer