nodemcu/nodemcu-firmware #1165: Simplify unnecessary prefixes in code

TerryE, 2016-03-17:
I raised this one in #873.
 
We have the `c_` prefixes for the string routines.  We have `malloc`, `c_malloc`, `os_malloc`, `mem_alloc`, etc. these are mostly synonyms  for their equivalents.  I personally would prefer the firmweware just to stick to the standard `strlen`, `malloc` etc.  In terms of mallocing we should only have a separate name if we want different functionality for example `luaM_malloc(L,t)` does a malloc but throws a Lua error if the malloc fails.  Most of these are define or linker synonyms.  It would really make working with the firmware base a lot simpler if we didn't have all of these redundant synonyms around the place.  If we actually have separate copies of routines, then this redundancy should be removed.  This is basically a bulk edit to map these down to the bare form.  I would prefer to do this as a one-off bulk edit and commit, but before we take this any further, I am interested in what the other contributors views are.  I've attached a full analysis FYI.

Function|(none)|c_|ets_|os_
-----------|---|---|---|----------
abs|2|3| | 
atoi|6|3||
bzero|1|||13
clearerr||2||
exit|14|4||1
fclose|2|4||
feof||3||
ferror|1|3||
fflush|2|6||
fgets||4||
fopen|2|4||
fprintf|9|6||
fputs||10||
fread||3||
free|34|83||62
freopen||3||
fscanf||2||
fseek|1|2||
ftell||2||
fwrite|1|2||
getc|1|6||
getenv|2|5||1
malloc|32|20||19
memcmp|4|13|1|
memcpy|11|86|1|
memmove|||1|
memset|22|61|1|
printf||31|3|
putchar||2||
puts||5||
rand||2||
realloc||3||
round||6||
setvbuf||2||
sprintf||49|2|
srand||2||
strcat||5||
strchr||22||
strcmp||45|1|
strcoll||3||
strcpy||10|1|
strcspn||3||
strerror||1||
strlen||74|1|
strncasecmp||2||
strncat||17||
strncmp||8|1|
strncpy||2|1|
strpbrk||3||
strrchr||2||
strstr||4|1|
strtod||6||
strtol||3||
strtoll||3||
strtoul||4||
tmpfile||2||
ungetc||3||
zalloc||32||

djphoenix, 2017-01-19:
I totally agree for use consistent naming, and agree for non-prefixed usage.

TerryE, 2018-02-10:
As far as the Lua core is concerned, I will be doing this in the NodeMCU Lua 5.3 release anyway.  I now see little point in doing this for legacy 5.1 Lua code.  There is separate discussion about whether we bother to tidy up or core modules, but this is really a subsidiary discussion to the ESP32/ESP8266 unification.

closed 2018-02-10 by TerryE