nodemcu/nodemcu-firmware #1166: Unprotected mallocs

TerryE, 2016-03-17:
This is a side effect of #1165, when I decided just to do a quick survey of malloc usage in the source.  OK running out of heap is usually fatal, but it's better to get some "Heap exhausted" error before reboot than to trundle on and then get a hardware addressing exception.  So I had a quick scan of all allocs to see which don't check for a NULL return.  Here is what I found:

File|Untested API calls
----|--------------
driver/key.c|1 × os_zalloc()
http/httpclient.c|4 × os_malloc() also esp_strdup() prints warning in debug mode but still uses pointer.
coap/node.c|1 × c_zalloc()
modules/mqtt.c|5 × c_zalloc() detects but leaks
modules/crypto.c|3 × c_malloc() 1 × os_malloc() 1 × os_zalloc()
modules/ws2812.c|1 × c_malloc()
modules/enduser_setup.c|1 × os_zalloc()
task/task.c|1 × os_malloc()
lwip/core/mdns.c|1 × os_zalloc()
lwip/app/espconn_udp.c|1 × os_zalloc()
lwip/app/dhcpserver.c|1 × os_zalloc()
lwip in general|This uses LWIP_ASSERT to check that allocation pointers are non zero but this is functional noop if LWIP_DEBUG is not defined.
app/cjson/dtoa.c|Lots × malloc()

The LWIP ones are particularly concerning.  Also note that one: `task/task.c` is my bad. :disappointed:  My only excuse is that it is part of system start up, and needed to run Lua so if this one fails then we are totally dead.  

Also of general concern is that many routines do some form of short circuit return which rolls up the stack and can get lost.  My view is that the built-in allocator is OK during initialisation but I think that we should really deprecate its use in modules for the Lua allocator which can trigger GC and retry or a proper throw of the error.

Again comments from contributors would be appreciated.

pjsg, 2016-03-17:
The problem with using the Lua allocator is that it longjumps on an error. This is difficult to catch in C code. This means that you are unable to do any cleanup if a function call fails (as you never get control back). However, the fact that it does GC is really handy. Maybe there is some sort of compromise.

TerryE, 2016-03-17:
@pjsg Philip I had real fun with the packed line info patch to the Lua core removing these types of leak that the Lua test suite exposed, so I understand this issue, but we need to balance risks here. In general running out of heap is fatal, so the C so developers needs to restructure their apps to avoid this excessive memory use.

But when the LWIP stack hits one of these exhaustions, it doesn't shut down with a "I am rebooting because memory is exhausted in LWIP. Goodbye!" message, the ESP just ends up doing a memory exception reboot, so the app programmer  doesn't lay the blame clearly on resource exhaustion, but thinks "oh there is some intermittent bug in the firmware".

How many times have you done code reviews where the the developer does something like `if (!p) return NULL; ` to "control " this type of error, and the layers up the stack frame the function has not worked quite right but has generated no meaningful diagnostic?

pjsg, 2016-03-17:
My issue with the longjump behavior of luaM_malloc is that it is effectively *impossible* to write code that continues after a memory allocation error. If we want to make these errors always cause a PANIC (with no possibility of trapping via xpcall), then that is one approach. The alternative is to have a call that can fail (after doing GC etc). 

Even if we pick the latter approach, the GC can move certain stuff around and so there may be places where there is an assumption that this stuff is not being moved around.

TerryE, 2016-03-17:
> My issue with the longjump behavior of luaM_malloc is that it is effectively impossible to write code that continues after a memory allocation error.

Well Lua itself manages well and as I said my packedLineInfo patch ditto.  I am confused about your xpcall comment because the whole point of this is to establish an error handler to catch any errors thrown.  What you can't do is resume at the point of the error, but how could you meaningfully do that anyway with an out of memory error?  The GC is integrated into luaM_malloc so that is will be a GC and retry before giving up.  

To me this is a 90:10 issue.  This gives a far more diagnosable outcome in 90% of the cases.  In 10% it can complicate the code, but if you can give me an example of the 10% then I'll show you how to code it.   I would make the use of luaM_alloc a *guideline* and not a *mandatory* anyway.  It is just that not using the Lua allocators would be a ***danger*** signal for reviewers to check that the out of memory errors were caught properly and processed at an application level. 

But this is very much a side issue / rabbit hole compared to the main thrust of the thead -- we've got uncaught malloc failures; what should we do with them?

jmattsson, 2016-03-18:
>  we've got uncaught malloc failures; what should we do with them?

Fix them? ;)

The lwip ones are problematic as we effectively replace that entire codebase on most SDK upgrades, which makes keeping local patches challenging. It also makes it a reason to shove the LWIP code out from the main code base and unpack it together with the SDK and patch it, but that's a fair chunk of extra effort right there too. It almost makes me suggest that we should stop using the open source version on LWIP and just suck up the stupid binary blob, but I disdain binary blobs...

As for using the Lua memory allocator, I think that should be encouraged in any of the module code, but leave libraries and platform things "raw". I think I've mentioned my preference on this elsewhere before.

Speaking of memory, I have a set of diagnostic wrappers for the memory allocation which tracks allocs/frees in order to provide leakage reports (and catch things like double-frees). Would that be of interest to the project? Essentially each module being instrumented gets its own tracking pool, and enabling/disabling is with a simple #define before including the header file. When disabled it all macros out to no overhead.

TerryE, 2016-03-18:
As far as LWIP goes, I have had another look, and these are largely OK in that only seems to use its own realloc to decrease a block size and assumes that a reducing realloc can never fail.  Nonetheless, I still think that running out of heap in LWIP will usually end messily.

I would prefer the respective maintainers to fix the module leaks.

pjsg, 2016-03-18:
As far as adding diagnostics to malloc/free, I am all for that. I had a version that would clear the memory to 0xAA before freeing it so that any references to freed memory would be obvious as well. Also, preset newly allocated memory to 0x55 so that uninitialized memory was more obvious. Also, make sure that realloc always moves the block. Also add a guard byte (or more) to each end of the memory block to look for overflows/underflows (and double frees).

jfollas, 2016-04-04:
Would it be too much to ask for a C Programming Standards for NodeMCU section to be added to the docs, containing examples of how and when (always?) to use `luaM_malloc` and gotchas (i.e., when the code returns before `free` is invoked, etc)?  Same for the various C functions (i.e., `c_*` vs `os_*`, what #include is needed, etc).

I know there are a lot of C resources just a Google search away, and most of us (?) took courses on C in university, but if there are ways that things should be implemented in this project, or if you're finding a lot of the same mistakes being made, then it would be good to have them listed in one place for reference.

I'm able to feel my way through a lot of tasks by referring to other code, but this isn't always the best way if they didn't follow any sort of standard (and as you showed above, not checking for NULL is a common practice in this codebase).

TerryE, 2016-04-04:
Note quite the Lua memory system doesn't return a NULL if an alloc fails, it throws an error so there's no point in testing for NULL, but yes we should have some guidelines.

TerryE, 2018-03-27:
We should really pick this up in #2303.  No point in keeping these two closely related issues open.

closed 2018-03-27 by TerryE