nodemcu/nodemcu-firmware #1171: gpio.trig stops working after awhile

Na2l, 2016-03-18:
```
state=0
function debounce (func)
  local last = 0
  local delay = 200000

  return function (...)
    local now = tmr.now()

    if gpio.read(switchPin) == 1 then -- 1 means up
      if now-last >= 3000000 then
        print ("pushed for " .. (now -last) )
      end
    end

    if now - last < delay then return end

    last = now
    if gpio.read(switchPin) == 0 then
      return func(...)
    end
  end
end
function toggle()
  if state == 0 then
    state = 1
    gpio.write(outputPin,gpio.HIGH)
  else
    state = 0
    gpio.write(outputPin,gpio.LOW)
  end
end

gpio.trig(switchPin, 'both', debounce(toggle))

```
this works just fine for a few hours, then pressing the button does not do anything. if I execute:
`gpio.trig(switchPin, 'both', debounce(toggle))`
it starts working again but then it stops after a few hours.

TerryE, 2016-03-18:
What firmware version are you using?

Na2l, 2016-03-18:
I believe it is based on SDK 1.4.0.
On March 2nd, I got the firmware from: http://nodemcu-build.com/ (master)
/N

> What firmware version are you using?
>
> ---
> https://github.com/nodemcu/nodemcu-firmware/issues/1171#issuecomment-198459292

/N

TerryE, 2016-03-18:
Then please retry with `dev` as this processing has been rewritten to sort out various issues.

closed 2016-03-19 by TerryE