nodemcu/nodemcu-firmware #1172: unsubscribe a mqtt topic

wuyuanyi135, 2016-03-19:
Hello,

The mqtt library does not provide unsubscribe feature. Will it be implemented?

marcelstoer, 2016-03-19:
We're always happy to see pull requests created by the community. It's true if there's `subscribe` you'd expect `unsubscribe` but I don't actually see the need for it. What's the use case?

pjsg, 2016-03-19:
Actually, the project that I am working on needs the unsubscribe functionality as well. I had assumed that it was present as well. 

My particular project is a dial which shows a value (such as electricity consumed/generated). It uses the rotary switch module (that I committed) to choose which parameter to display. It then subscribes to that parameter and then drives a stepper motor to drive the needle (once we get past the current dev->master merge, I'll submit that). Of course, I need to unsubscribe the old topic.....

marcelstoer, 2016-03-28:
@pjsg thanks for the PR

PR #1179 (2016-03-28): MQTT Unsubscribe support
This is the change to support unsubscribe in MQTT that is compatible with the subscribe method,

Tested to subscribe and unsubscribe from a single topic and multiple topics.

This should resolve #1172
closed 2016-03-28 by marcelstoer