nodemcu/nodemcu-firmware #1173: wifi.startsmart broken in latest dev

danlu01, 2016-03-19:
Just flashed the latest dev firmware and wifi.startsmart is broken.

Using example startsmart code in doc:
`wifi.setmode(wifi.STATION)
wifi.startsmart(0,
    function(ssid, password)
        print(string.format("Success. SSID:%s ; PASSWORD:%s", ssid, password))
    end
)`

Here's the result:
`stdin:1: attempt to call field 'startsmart' (a nil value)
stack traceback:
	stdin:1: in main chunk
`

Looks like startsmart was removed?? Was there's any updates to startsmart?

TerryE, 2016-03-19:
It has a heavy runtime footprint, and few users use the API so it is now disabled by default, but can be enabled via `user_config.h` setting.

danlu01, 2016-03-19:
I use the cloud firmware build from http://nodemcu-build.com/index.php.  How do I enable it with that?

marcelstoer, 2016-03-19:
> Looks like startsmart was removed?

Yes, see #1104

TerryE, 2016-03-19:
The cloud builder only offers limited configuration options. Our documentation explains that you can quite easily configure your own container build on your device PC.

marcelstoer, 2016-03-20:
> wifi.startsmart broken in latest dev

As stated it ain't *broken* it's simply disabled by default -> closing issue.

closed 2016-03-20 by marcelstoer