nodemcu/nodemcu-firmware #1174: LCD (128x128) with ILI9163 not updated correctly through ucg

FrankX0, 2016-03-19:
I have an LCD (128x128) with ILI9163 controller connected to through SPI.
When initialising the screen, part of the screen is not updated (random data).
I found the related issue on the ucglib repository:
https://github.com/olikraus/ucglib/issues/60
I can confirm this solves this issue.

marcelstoer, 2016-03-20:
> I can confirm this solves this issue.

Great, what's holding you back from submitting a PR then?

@devsaurus Arnin, would it even make sense to fix Ucglib issues in NodeMCU directly? After all, what you integrated is based on the latest available release (v1.3.3). Ideally all issues are fixed upstream and you simply pull in new releases but what is the *realistic* process to handle issues like this one?

pjsg, 2016-03-20:
My take is that we *really* want to take the upstream fix directly rather than letting the code bases diverge.

marcelstoer, 2016-03-21:
Of course I fully agree with you Philip. I'd be really odd if NodeMCU Ucglib had fewer bugs than the original Ucglib. 
On the other hand if someone created a PR for this issue here he/she could submit the same for @olikraus upstream. The fix(es) wouldn't be pulled into the repository at the same but the code bases wouldn't actually diverge. Once we update Ucglib we would be sure that our fix wouldn't be undone.

devsaurus, 2016-03-21:
I agree with the statements from @marcelstoer and @pjsg.

Updates will be integrated into the NodeMCU firmware as soon as there's a new upstream release. At the moment we're at [Ucglib v1.3.3](https://github.com/olikraus/Ucglib_Arduino/releases/tag/v1.3.3).
I also can't investigate this issue since I don't have the specific ILI9163 variant on my desk. This applies for most of the supported display types, thus I'm aiming to ensure a clean integration in NodeMCU, but need to rely almost completely on the upstream verification in Ucglib (and U8glib).

closed 2016-03-21 by devsaurus