nodemcu/nodemcu-firmware #1178: OLED with SSD1306 giving random dots

maartenbreddels, 2016-03-20:
Bought this oled:
http://www.ebay.nl/itm/0-96-I2C-IIC-Serial-128X64-OLED-LCD-Display-Module-Arduino-Scientific-Device-/361028753804?
In nodemcu I cannot get it this OLED to work, it gives the random dots (or noise) many people complain about. I feared it had to do with the reset issue, but it works perfect with espeasy.
I noticed the initialization is a bit different, 
https://github.com/ESP8266nu/ESPEasy/blob/45571e0b06a80509fe35f467d800612453446adf/_P023_OLED.ino
I don't have time atm to try to fix this, but maybe someone finds this useful info for a PR.

UncleRedz, 2016-03-21:
I'm running something that looks the same, same controller and size of OLED. I get the random noise on first initialization after powering on. Once I draw something, the display is cleared and shows what it should. Doing a reset of the ESP while keeping power on, keeps the display as it was before resetting. Again, after first drawing it shows what it should. 

It's like it's not doing a clear of the display buffer when initializing the display. Apart from this, it works nicely. 

Is this what you are seeing as well? Or do you have random noise all the time when drawing?

devsaurus, 2016-03-21:
Thanks for your confirmation @UncleRedz.

@maartenbreddels clearing the display has to be done from the Lua script. Can you please post your Lua code here to investigate potential integration issues?

maartenbreddels, 2016-03-22:
I only was able to get noise, never got to display anything. I was playing with this snippet below using ESPlorer, I left the code as it was (with comment, so you know what I was trying). I also tried powering the OLED on and off, and resetting the esp, but nothing worked. Except that the first few lines (or bottom ones?) seems to be cleared.
```
id=0
sda=5
scl=6
addr=0x3c

i2c.setup(id,sda,scl,i2c.SLOW)
disp = u8g.ssd1306_128x64_i2c(addr)
--i2c.setup(id,sda,scl,i2c.SLOW)
disp:setFont(u8g.font_6x10)
disp:begin()
disp:firstPage()
--print(disp)
--disp:setColorIndex(1)
disp:drawBox(5, 10, 20, 10)
disp:drawBox(10, 20, 20, 20)
disp:drawStr(0, 20, "Blaat")
--disp:fillBox(0, 0, 50, 50)
disp:nextPage()
```

UncleRedz, 2016-03-22:
Sounds like it's working, you just need to draw the display multiple times until there are no more pages. That is how U8glib works, it splits the display into multiple pages or slices, which you draw one at a time. Read the u8glib documents and also check the example code in this repository. (Keep in mind that the example lua code for u8glib is not good as it draws all pages in one go, which will cause the Wifi to timeout.)

TerryE, 2016-03-22:
@UncleRedz 

> Keep in mind that the example lua code for u8glib is not good as it draws all pages in one go, which will cause the Wifi to timeout.

Care to offer a better implementation? :smile:

maartenbreddels, 2016-03-22:
Ok, feeling stupid now :), it is working. Didn't realize u8glib worked that way, thx for the replies.

closed 2016-03-22 by maartenbreddels