nodemcu/nodemcu-firmware #1187: Customizable Enduser-Setup webpage

skycoders, 2016-03-25:
I just tried the enduser setup module for the first time. It worked great. Nevertheless, I was wondering if anyone agrees that handing over a file name from where to load the html file in the start call would be cool. The html page could be provided as a template in git or wherever. This file would then replace the website inlined into the c module file. Any comments?

robertfoss, 2016-03-25:
That feature already exists. Place a file named index.html in the filesystem, and it will be served by EUS.

skycoders, 2016-03-25:
uh, sorry... but that is not documented in the module docs, isn't it? I could find that in the module code now. 
Besides this, changing the AP name from SetupGadget... to something else is not included, correct? What do you think about this extension?

robertfoss, 2016-03-25:
It's possible in manual mode or while compiling NodeMCU.

closed 2016-03-25 by marcelstoer

charudatt, 2016-08-28:
Hi Robert. 
I tried your code along with a non customized bin file and it worked the first time, but again it just goes in a loop, It does not connect to the accesspoint nor does it show up the HTML page.

When I used the compiled NODE MCU bin file, the NodeMCU board just goes into a endless loop showing junk characters in the serial screen.

Can you please explain how to use it with an example.

robertfoss, 2016-08-28:
@charudatt this isn't the place for this discussion. hit me up at dev@memcpy.io

Buuut, have a look at building a module with @marcelstoer fantastic service: https://nodemcu-build.com/
Build a module with the enduser_setup module compiled in.

Run a an init.lua file with something along the lines of:

    function on_connect()
      print("Connected to wifi as: " .. wifi.sta.getip())
      ssid,password,bssid_set,bssid = wifi.sta.getconfig()
      print(
          "\nCurrent Station configuration:"
        .."\nSSID : "..ssid
        .."\nPassword  : "..password
        .."\nBSSID_set  : "..bssid_set
        .."\nBSSID: "..bssid.."\n"
      )
      -- add more stuff here
    end
    
    enduser_setup.start(on_connect)`

marcelstoer, 2016-08-28:
> this isn't the place for this discussion. hit me up at dev@memcpy.io

True, but I'd much rather if you guys could discuss in public so others could profit as well 😉 @charudatt also see https://nodemcu.readthedocs.io/en/latest/en/support/

gusarov, 2016-11-19:
Sorry, is there a way to call and configure enduser-setup on C (using Visual Studio / Atmel Studio / Arduino Studio)? I like ESP & Nodemcu but have no intention to learn one more new language.

robertfoss, 2016-11-19:
The module is written in C.

So you can do whatever you like with it in your editor of choice.