nodemcu/nodemcu-firmware #1188: ssd1331_18x96x64_univision_hw_spi crash

asantoni, 2016-03-25:
There's something funky going on with the ucg.ssd1331_18x96x64_univision_hw_spi module that causes nodemcu to crash. I demonstrate this below in my code snippets taken from the interactive lua console over a serial port.

If I turn on the relevant #define for it in include/ucg_config.h, the ucg.ssd1331_18x96x64_univision_hw_spi function exists but is nil somehow. 

```
=ucg.ssd1331_18x96x64_univision_hw_spi(1, 2, 0)
stdin:1: attempt to call field 'ssd1331_18x96x64_univision_hw_spi' (a nil value)
stack traceback:
	stdin:1: in main chunk
```

So here's the extra weird part - If I try to enumerate over the ucg module, I get a crash:

```
> function test()
>>    for key, value in pairs(ucg) do
>>        print(key)
>>        print(value)
>>    end
>> end
> =test()
=test()
ssd1331_18x96x64_univision_hw_spi
lightfunction: 0x40263e88
-2308845

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30188, room 16 
tail 12
chksum 0x9f
ho 0 tail 12 room 4
load 0x3ffe8000, len 2208, room 12 
tail 4
chksum 0x1e
load 0x3ffe88a0, len 8, room 4 
tail 4
chksum 0xdb
csum 0xdb
don't use rtc mem data
rlŽ‚rlŒžâŒbþ

NodeMCU 1.5.1 build unspecified powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
lua: cannot open init.lua
```

For comparison, if I compile just the st7735 module in, I can run the test function with no problems:

```
> function test()
>>    for key, value in pairs(ucg) do
>>        print(key)
>>        print(value)
>>    end
>> end
> =test()
st7735_18x128x160_hw_spi
lightfunction: 0x40263e88
font_7x13B_tr
userdata: 0x40272629
font_helvB08_hr
userdata: 0x40272155
FONT_MODE_TRANSPARENT
1
FONT_MODE_SOLID
0
DRAW_UPPER_RIGHT
1
DRAW_UPPER_LEFT
2
DRAW_LOWER_RIGHT
8
DRAW_LOWER_LEFT
4
DRAW_ALL
15
__metatable
romtable: 0x4026ff20
> 
```

The map is corrupted or something?

Any help or tips would be greatly appreciated. Thanks!

devsaurus, 2016-03-25:
I can confirm your observations, thanks for reporting.

It all boils down to the fact that `ssd1331_18x96x64_univision_hw_spi` as a string is 33 characters long. According to `lrotable.h` the maximum name length is 32, and this ends up to be the root cause for the problem you encountered.
I'm going to fix that soon.

asantoni, 2016-03-25:
That was it! I shorted that symbol (1 line change) and everything started working as expected. :+1: 
[image]

devsaurus, 2016-03-25:
:smile: excellent :+1:

TerryE, 2016-03-26:
Hummn. This fix is good, but what is a little concerning is that this should fall silently in the first place. Personally, I would prefer to error to be raised during startup so that this issue can be picked up during module development and testing.

PR #1191 (2016-03-25): Shorten ucg constructors to keep them below LUA_MAX_ROTABLE_NAME
Fixes #1188.
closed 2016-03-26 by marcelstoer