nodemcu/nodemcu-firmware #1209: Dev Branch - Can't autodetect firmware, because proper answer not received.

LostLobos, 2016-03-31:
I was attempting to use the nodeMCU from the Dev Branch but got no success.

I've builded all modules from the Dev branch using this tool: http://frightanic.com/nodemcu-custom-build/

The firmware, using the nodeMCU flasher was loaded correctly on ESP8266. But when I try to run the firmware on the IDE **ESplorer** I receive this message:

`Can't autodetect firmware, because proper answer not received.`

Using the latest master I had no problems.

pjsg, 2016-03-31:
I suspect that this is because the baud rate has been changed to 115200 for the dev branch.

LostLobos, 2016-03-31:
Thank you @pjsg. It changed to this value and it worked like a charm.

I was with this problem for days, really appreciated!

closed 2016-03-31 by LostLobos

marcelstoer, 2016-03-31:
> I was with this problem for days

Reading the docs would have helped - if you know what to look for: http://nodemcu.readthedocs.org/en/dev/en/upload/

pjsg, 2016-03-31:
I think that this does point out a serious issue is that after the master merge, everybody is going to face this problem.......

marcelstoer, 2016-03-31:
Yep, was bound to happen sooner or later. It's on the [README](https://github.com/nodemcu/nodemcu-firmware/tree/dev#set-uart-bit-rate), it's in the [docs](http://nodemcu.readthedocs.org/en/dev/en/upload/) (more than one place), the docs are linked on README but...we know people don't read documentation. I'll also update the front page of my build service and the documentation of my Docker image once the master merge has happened to mitigate the problem.

marcelstoer, 2016-03-31:
Oh, and fixing #1077 would of course help, too. We could then also add such info to the template.

LostLobos, 2016-03-31:
I really sorry for the Trouble. I looked at Flashing Firmware twitce but didn't looked into Uploading Code ( The answer was right in my face ).

When merging the dev branch with master consider putting this information on top of Readme. I didn't know what I was looking for or what was my problem. You could reference this Issue on the page too, kinda like a FAQ to prevent repetitive questions.

kawsar2223, 2016-08-13:
[image]
[image]

marcelstoer, 2016-08-13:
@kawsar2223 
- you're using 0.9.6 which is no longer maintained, use a recent version https://github.com/nodemcu/nodemcu-firmware/releases
- then consider http://nodemcu.readthedocs.io/en/latest/en/support/ and #1010