nodemcu/nodemcu-firmware #1217: MQTT callback function with QoS > 0 problem

ricklettow, 2016-04-02:
Posted an example of the issue here as well -> http://www.esp8266.com/viewtopic.php?f=18&t=9038&p=44221#p44221

There is a problem with the callback function reference in the publish and subscribe functions of the mqtt module.  If you publish more than one message with qos>0, the callback function is always overwritten by the last publish() command.  This is a problem if you want a specific callback fired for each unique publish/subscribe command.

**Disclaimer: Beginner level, I am making my best guess to identify where the problem exists in the source code, hopefully someone with more experience will take a look.  Far from a fix but hopefully in the right direction**

branch: main, svn revision 4908
mqtt.c:line 1278
```
 if(mud->cb_puback_ref != LUA_NOREF)
      luaL_unref(L, LUA_REGISTRYINDEX, mud->cb_puback_ref);
    mud->cb_puback_ref = luaL_ref(L, LUA_REGISTRYINDEX)_
```
The message callback function should be part of the message object, not lmqtt_userdata?

mqtt_msg.h:line 68
```
typedef struct mqtt_message
{
  uint8_t* data;
  uint16_t length;

} mqtt_message_t;
```
Add callback function reference here?????

Other thoughts:
The message(with callback function) would have to remain in queue until it is successfully acknowledged, increasing necessary memory.  Cleanup of messages that fails to acknowledge?

I understand the callback function working as it currently does for qos 0 messages as there is no way to verify the unique message but it may be worth indicating this in the mqtt module publish() and subscribe() command documentation.

pjsg, 2016-04-02:
Ah yes -- I can believe this entirely. I think that the issue still happens with qos 0 messages as they are internally queued. 

Do you want to do a PR to fix the documentation to indicate this behavior. It'd be nice to get the doc change in before the pending merge-to-master. Any fix could follow on (and would update the documentation again).

ricklettow, 2016-04-02:
You are correct on the qos 0 messages as well, I had not thought it through completely...

I agree with the need to update the documentation until the issue is resolved and will attempt a PR (my first one so please forgive me if i mess it up).

pjsg, 2016-04-02:
I wonder how many other modules in nodemcu suffer from the same problem.....

ricklettow, 2016-04-02:
Before I attempt the PR, does this wording make sense?

**mqtt.md......**
#### Parameters
- `topic` the topic to publish to ([topic string](http://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices))
- `message` the message to publish, (buffer or string)
- `qos` QoS level
- `retain` retain flag
- `function(client)` optional callback fired when PUBACK received. 
NOTE: When calling publish() more than once, the last callback function defined will be called for ALL publish commands.

ricklettow, 2016-04-02:
PR 1218.  Please look it over.

ricklettow, 2016-04-02:
Thank you @pjsg.  I will attempt to fix but don't hold your breath.

closed 2019-06-21 by stale[bot]