nodemcu/nodemcu-firmware #1220: Change enduser_setup connected callback to fire when enduser_setup is done

pjsg, 2016-04-03:
I want to be able to write code like:

init.lua:
```
enduser_setup.start(function() dofile("init2.lua") end)
```
init2.lua:
```
-- start the http server
-- start the timers
-- etc, etc
```

However, the connected callback currently fires while the enduser_setup still has the http server active so I can't start another one. Adding a timer for 10 seconds seems the wrong approach as well.

Having to wrap all my initialization code in a pcall and then retrying it every second also seems the wrong approach, -- and it would require very careful coding to make sure that everything only gets initialized once.

Please have a callback that indicates when the wifi is working and the enduser_setup module has cleaned up after itself.

jfollas, 2016-04-03:
As a workaround, you should be able to call `enduser_setup.stop()` within the callback to tear down EUS before the 10-seconds elapse.

jmattsson, 2016-04-15:
And if that's causing you crashes, use manual mode (you'll need to get the ESP into STATIONAP mode before starting EUS in that case).

It does sound like the callback is firing too early as is though.

closed 2019-06-21 by stale[bot]