nodemcu/nodemcu-firmware #1225: Memory leak when stuck in STATION_NO_AP_FOUND

karrots, 2016-04-05:
The below code causes a very quick memory leak when the module is stopped in `STATION_NO_AP_FOUND` mode. Only going into dsleep with option 4 seems to cause the memory leak.

```
wifi.setmode(wifi.STATION);
wifi.setphymode(wifi.PHYMODE_N);
wifi.sta.config("<workingSSID>","mypassphrase",1);

-- =wifi.sta.status() shows module getting to status `STATION_GOT_IP`
node.dsleep(1000000,4);
-- =wifi.sta.status() shows module getting stopping at  `STATION_NO_AP_FOUND` which is to be expected. Heap goes down rapidly and reboots the module.
```

Testing with a bad SSID does not exhibit the same issue while stopped in the `STATION_GOT_IP`.

dnc40085, 2016-04-10:
As a workaround you could make sure you run `wifi.sta.disconnect()` before running `node.dsleep(1000000,4)`

karrots, 2016-04-10:
True, I'll keep that in mind if I end up using level 4 sleep in my project. For now, I was just doing current draw tests with the different options and found the issue. 

I should clarify my last sentence where I say "bad" SSID. In this test case, the unit was never put to sleep I just provided it a bad SSID to test if the leak was unique to `dsleep` level 4 or `STATION_GOT_IP`.

closed 2019-06-21 by stale[bot]