nodemcu/nodemcu-firmware #1228: Add system_get_time() as extra argument to gpio.trig callback

TerryE, 2016-04-09:
Some gpio codes need to know the best estimate of the triigger time.  I suggest that we add some code to pass the `system_get_time()` back through the task post to the trigger callback handler and pass this time to the callback as an extra argument.  This way we maintain backwards compatibility with the existing API and allow a lot of time-critical code with used to work under the old nasty ISR trigger to be recoded and work.

Note at we can only pass back a 32 bit argument and we need 6 bits for the level and pin, but that still leaves the 26 lsb of the system time, but since we can assume that tasks will be delivered in less than 30s, say, then we can recover the full system time.

This is a simple patch if there's agreement to do this.

pjsg, 2016-04-09:
I'd like to leave a little more space for the pin number -- at least one more bit (i.e. 6 -- 0..63). The reason is that I'm looking at refactoring the gpio code so that you can add i2c gpio extenders, and a 16 gpio port expander might have pin numbers 16 .. 31 etc. Essentially this is just adding an extra layer of indirection between the gpio module and the platform code and then having a registration mechanism for other drivers to plug in.

I do support passing the time back.....

TerryE, 2016-04-09:
2^25 µSec is 33s. so if we pass back the 25 lsb, the delivery latency is typically 10-30 µSec so something would have to be pathologically wrong to select the wrong epoch.  If the latency is > 5s then something is *seriously* wrong and the system shouldn't fail silently, I suggest that we return a nil if the time is greater than that, so this can be detected programmatically.

jmattsson, 2016-04-15:
Is this the point where I point out that ESP time keeping is not up to scratch, and some of us would prefer to use the rtctime time instead of SDK time? :)

TerryE, 2016-04-15:
I agree that this might be a case of an optional define `GPIO_USE RTCTIME` or `GPIO_USE_SDKTIME`.  The real purpose of this is offsetting for the delta between ISR delivery and the start of the task which is normally a few 10s of microseconds if the chip is idling at ISR or 10s of  mSec if another task is running.  For this type of offset the SDK time I'd usually good enough, IMO.

jmattsson, 2016-04-16:
Yeah I agree it wouldn't be all that hard to take an SDK time delta and apply to RTC time to get what I want. I wouldn't say no to a config option to get the RTC time domain directly though. Quite low prio, especially since I don't need it right now.

pjsg, 2017-01-02:
This was done as part of some merge...

closed 2017-01-02 by pjsg