nodemcu/nodemcu-firmware #1229: wifi.STATION not start while GPIO0 or GPIO2 are connected

guilhermenicolini, 2016-04-10:
In ESP8266 ESP-01 I load the following code:

wifi.sta.setip({ip="192.168.0.109",netmask="255.255.255.0",gateway="192.168.0.1"})
wifi.setmode(wifi.STATION)
wifi.sta.config("xxxxxx","xxxxxxx")

led1 = 3
led2 = 4

gpio.mode(led1, gpio.OUTPUT)
gpio.mode(led2, gpio.OUTPUT)

srv=net.createServer(net.TCP)

srv:listen(80,function(conn)
    conn:on("receive", function(client,request)
        local buf = "";
        local _, _, method, path, vars = string.find(request, "([A-Z]+) (.+)?(.+) HTTP");
        if(method == nil)then
            _, _, method, path = string.find(request, "([A-Z]+) (.+) HTTP");
        end
        local _GET = {}
        if (vars ~= nil)then
            for k, v in string.gmatch(vars, "(%w+)=(%w+)&*") do
                _GET[k] = v
            end
        end
        buf = buf.."HTTP/1.1 200 OK\r\n";
        buf = buf.."Content-type: text/html\r\n";
        buf = buf.."Connection: close\r\n\r\n";
        buf = buf.."<!DOCTYPE html>";
        buf = buf.."< html >";
        buf = buf.."< body >";
        buf = buf.."< hi > ESP8266 Web Server< /h1 >";
        buf = buf.."< p >GPIO0 < a href=\"?pin=ON1\" >< button >ON< /button >< /a >&nbsp;< a href=\"?pin=OFF1\" >< button >OFF< /button >< /a >< /p >";
        buf = buf.."< p >GPIO2 < a href=\"?pin=ON2\" >< button >ON< /button >< /a >&nbsp;< a href=\"?pin=OFF2\" >< button >OFF< /button >< /a >< /p >";
        buf = buf.."< /body >";
        buf = buf.."< /html >";
        local _on,_off = "",""
        if(_GET.pin == "ON1")then
              gpio.write(led1, gpio.HIGH);
        elseif(_GET.pin == "OFF1")then
              gpio.write(led1, gpio.LOW);
        elseif(_GET.pin == "ON2")then
              gpio.write(led2, gpio.HIGH);
        elseif(_GET.pin == "OFF2")then
              gpio.write(led2, gpio.LOW);
        end
        client:send(buf);
        client:close();
        collectgarbage();
    end)
end)

My module is powered with FTDI232 in 3.3v pin with the following ports:

FTDI232 | ESP-01
------------ | -------------
VCC | VCC
VCC | CH_PD
GND | GND

When I connect the GPIO 0 and 2 in the LEDs and turn on the power, all LEDs is HIGH and the server do not start.
When I disconnect the GPIO ports and turn on the server starts. Then if I connect the GPIOs the code works normally

sekarpdkt, 2016-04-10:
Did you try to keep CH_PD floating. Normally for uploading script, it need not be connected to VCC

TerryE, 2016-04-10:
@guilhermenicolini, this isn't really an issue with the firmware, but more a how do I configure my hardware +fix the bugs in my code. See #719. You need to ask this sort of Q on a programmer forum.

closed 2016-04-10 by TerryE

guilhermenicolini, 2016-04-10:
Thanks. I ll use this forums. Just sure that is not a bug in the load code.