nodemcu/nodemcu-firmware #1230: UDP Broadcase receive is not working

sekarpdkt, 2016-04-10:
Hi UDP broadccast receive is not working. Have done 2-3 days of google and not finding any solution.

-- Broadcast from both ESP8266 in same network is received by node.js in pi
-- Broadcast is not received by other module in same network (both modules are broadcasting)
-- both modules does not receive broadcast from pi
-- Raspberry pi is acting as AP/Gateway
-- Broadcast by pi is received by pi itself (node.js) 

`
port=8972
Srv=net.createServer(net.UDP)
Srv:on("receive", function(Srv, pl)
      print("received :"..pl)
      Srv:send("Yes") 
   end
)
Srv:listen(port)

tmr.alarm(0, 60000, 1, function()
	client=net.createConnection(net.UDP)
	client:connect(8972, "192.168.42.255")
	client:send("hi")
	client:close()
end
)

`

TerryE, 2016-04-10:
@sekarpdkt, please use markdown to format your Qs correctly so that others can read them easily. If you want answers then please show this courtesy in return. But I will look at this broadcast issue.

sekarpdkt, 2016-04-10:
Apologies. I will take care. (Acutally learning to use github)..
Meanwhile, I kept the setup on for few hours and slept. Interestingly, I am observing one module had received few broadcast messages, not all.. **Intermittent**.. Not sure...

received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi
received :hi

>

sekarpdkt, 2016-04-10:
I setup some proper wifi AP and tried broadcast using that (instead of AP setup in Raspberry PI) and it worked without any issues... All broadcast packets were received ESP8266. Not yet tried to connect Raspberry PI to that network, but closing this **as it looks like some other problem,** not related with ESP8266/NodeMCU.

closed 2016-04-10 by sekarpdkt

sekarpdkt, 2016-04-13:
connected Raspberry to that network and all worked fine as expected. Just to keep this updated, in case some one reaches thru google search... if I connect to AP hosted by Raspberry Pi, it is not working, but working ok with normal standard AP like D-Link DIR 505.