nodemcu/nodemcu-firmware #1232: UDP server on("receive") callback not cleared after close

TerryE, 2016-04-10:
```Lua
tmr.alarm(1, 5000, 1, function(check)
  i = (i or 0) + 1
  print( wifi.sta.getip())
  if (i==1) then
    srv=net.createServer(net.UDP)
    srv:listen(8888)
    srv:on("receive",  function(...) print ("hello") print(...) end)
  elseif (i==2) then
    srv:close()
  elseif (i==3) then
    tmr.unregister(1)
    for _,v in pairs(debug.getregistry()) do 
      if type(v) == "function"  then v("there") end
    end
  end
end)
```
prints the getIP output three times followed by the `hello` and `there` lines: the `on("receive")` is still registered.

marcelstoer, 2016-04-10:
#230 is related if not a complete dup.

TerryE, 2016-04-11:
It turns out for some reason the the destructor isn't called until the `__gc` method is fired which occurs when the socket goes out of scope or is set to nil -- unless of course the programmer has created a circular  upvalue reference. I will double check is a mo but in the above example setting `src = nil` after the close will trigger GC of the function unless I change the print to `print(srv, rec)`.  I will be on a test rig in an hour or so, so can check then.

jmattsson, 2016-04-15:
Is it just me or are we getting a lot of circular things in the net module which is gumming up gc? And if so, is that something you expect will be resolved by the big cleanup/rewrite of net?

TerryE, 2016-04-15:
We certainly need to get a handle on the limitations and flaws of the espconn layer and and come to an informed decision on the best approach.  However if we do decided to dump espconn, then I feel that we need to enforce it's removal from any modules.

djphoenix, 2017-01-19:
Tested on latest `dev`.
Callback not freed after `close` but cleaned up after `srv = nil`, so I think this is a normal behavior (`udpsocket` may be re-opened by `close->listen`, keeping all internal state with callbacks).

I think this may be closed. @TerryE agree?

TerryE, 2017-01-19:
@djphoenix Yuri, I am happy to close this.  The issue that bites a lot of Lua developers is that they don't understand some of the lazy GC characteristics of the language.  A classic one is leaving `srv` as an upval, either intentionally or accidentally.  The reference isn't GCed until (one of) the routines using it in upval scope sets it to nil or it/they are removed from active scope.

So this behaves as a proficient developer should expect but it is still a "bear trap" for typical ESP8266 Lua developers.  I don't think that we can easily solve this one.

closed 2017-01-19 by djphoenix