nodemcu/nodemcu-firmware #1236: mDNS: response sent to source instead of multicast address

FrankX0, 2016-04-11:
Thanks to @pjsg the mDNS functionality works better than before. But I found there is still room to improve.

When a request is sent to the multicast address 224.0.0.251, the node should send the response to the multicast address. Currently the response is sent to the source address.
The issue is in the nodemcu_mdns.c file, function mdns_recv.
When changing all the destination addresses to &multicast_addr, the conversion from a .local name to an IP address now works correctly for most browsers (Chrome, Safari, Firefox), ping and dns-sd.
In my opinion the regularly sending (currently every 120 seconds) of ms_info is no longer required (tested by disabling this functionality).
Please note that MS Edge is still not satisfied with the response. I'll try to find out why.

I would have liked to implement this change myself in the repository, but do not yet feel comfortably enough to do so.

TerryE, 2016-04-12:
>.I would have liked to implement this change myself in the repository, but do not yet feel comfortably enough to do so.

There is always the lesser option of proposing a patch if you've worked out a solution.

FrankX0, 2016-04-12:
I implemented following patch in _nodemcu_mdns.c_, in function _mdns_recv_, changing the source address of the mdns send calls from _addr_ to _&multicast_addr_
So, from:
`mdns_send_service_type(i, addr, port);`
to
`mdns_send_service_type(i, &multicast_addr, port);`
for all occurrences of _mdns_send_service_type()_ and _mdns_send_service()_.

pjsg, 2016-04-13:
You should only send the response to the multicast address if it is a QM query. 

I filed a PR to resolve this....

FrankX0, 2016-04-13:
Resolved by #1239 
Thanks @pjsg!

closed 2016-04-13 by FrankX0