nodemcu/nodemcu-firmware #1237: wifi.ap.config and wifi.sta.config signature

goliatone, 2016-04-12:
Any reasons why the signatures for `wifi.ap.config` and `wifi.sta.config` differ? 
`wifi.sta.config` takes multiple arguments and `wifi.ap.config` takes a table- `wifi.ap.dhcp.config` will also take a table.

jmattsson, 2016-04-15:
None that I'm aware of. I wouldn't mind seeing a patch to allow wifi.sta.config() take a table as its only argument (but we'd need to keep the current approach around for backwards compatibility for some time though).

eku, 2016-04-19:
Is there a way to mark API functions as deprecated?

marcelstoer, 2016-04-19:
We have two of those in the docs, `node.led()` and `node.key()`: http://nodemcu.readthedocs.org/en/dev/en/modules/node/#nodekey-deprecated

dnc40085, 2016-05-02:
@jmattsson I'll add this to my list of stuff to do.

On the subject of depreciation, I was thinking maybe we could add a `c_printf("this feature/parameter will be removed in version X.x, please update your code")` to the depreciated functions or parameters to let the user know that the old version is being phased out.

marcelstoer, 2016-05-02:
@dnc40085 that idea is ok BUT if we actually *had* releases to refer to we could as well add meaningful deprecation notes to the docs.

dnc40085, 2016-08-30:
@marcelstoer I'm currently writing a PR that will include this change, how should I handle the documentation change? 
Should I replace the old documentation entirely or should I keep the old documentation and add new entry for the new description?

marcelstoer, 2016-08-30:
Since the docs are in the repo alongside the code you can replace the current description so that it matches the new code.

dnc40085, 2016-11-07:
@marcelstoer since PR #1497 has been merged, I believe this issue can be closed.

closed 2016-11-07 by marcelstoer