nodemcu/nodemcu-firmware #1238: Error compiling lua cross - Ubuntu 14.04

ernitron, 2016-04-12:
I was trying to compile lua.cross but I got some errors compiling lua.cross as specified in docs. 

My environment is ubuntu 14.04

Linux dell 3.13.0-85-generic #129-Ubuntu SMP Thu Mar 17 20:50:15 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux. All the tool chain is ok and development environment is ok.

The compilation of nodemcu firmware was flawlessy. The compilation of lua.cross ended with errors.

This is the output of compilation is
`nodemcu-firmware# sudo lua tools/cross-lua.lua
[COMPILE] .build/cross-lua/app__lua__lapi.o
[COMPILE] .build/cross-lua/app__lua__lauxlib.o
[COMPILE] .build/cross-lua/app__lua__lbaselib.o
[COMPILE] .build/cross-lua/app__lua__lcode.o
[COMPILE] .build/cross-lua/app__lua__ldblib.o
[COMPILE] .build/cross-lua/app__lua__ldebug.o
[COMPILE] .build/cross-lua/app__lua__ldo.o
[COMPILE] .build/cross-lua/app__lua__ldump.o
[COMPILE] .build/cross-lua/app__lua__lfunc.o
[COMPILE] .build/cross-lua/app__lua__lgc.o
[COMPILE] .build/cross-lua/app__lua__llex.o
[COMPILE] .build/cross-lua/app__lua__lmathlib.o
[COMPILE] .build/cross-lua/app__lua__lmem.o
[COMPILE] .build/cross-lua/app__lua__loadlib.o
[COMPILE] .build/cross-lua/app__lua__lobject.o
[COMPILE] .build/cross-lua/app__lua__lopcodes.o
[COMPILE] .build/cross-lua/app__lua__lparser.o
[COMPILE] .build/cross-lua/app__lua__lrotable.o
[COMPILE] .build/cross-lua/app__lua__lstate.o
[COMPILE] .build/cross-lua/app__lua__lstring.o
[COMPILE] .build/cross-lua/app__lua__lstrlib.o
[COMPILE] .build/cross-lua/app__lua__ltable.o
[COMPILE] .build/cross-lua/app__lua__ltablib.o
[COMPILE] .build/cross-lua/app__lua__ltm.o
[COMPILE] .build/cross-lua/app__lua__lundump.o
[COMPILE] .build/cross-lua/app__lua__lvm.o
[COMPILE] .build/cross-lua/app__lua__lzio.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__luac.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__loslib.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__print.o
[COMPILE] .build/cross-lua/app__lua__..__modules__linit.o
[COMPILE] .build/cross-lua/app__lua__..__libc__c_stdlib.o
[LINK] luac.cross
.build/cross-lua/app__lua__luac_cross__loslib.o: In function `os_tmpname':
loslib.c:(.text+0x29c): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
.build/cross-lua/app__lua__lrotable.o: In function `luaR_findglobal':
lrotable.c:(.text+0x15e): undefined reference to `lua_rotable'
lrotable.c:(.text+0x17c): undefined reference to `lua_rotable'
lrotable.c:(.text+0x1b0): undefined reference to `lua_rotable'
.build/cross-lua/app__lua__..__modules__linit.o: In function `luaL_openlibs':
linit.c:(.text+0x6): undefined reference to `lua_libs'
linit.c:(.text+0x11): undefined reference to `lua_libs'
collect2: error: ld returned 1 exit status
[builder] Error building target
[builder] Last executed command was: 
gcc -o luac.cross .build/cross-lua/app__lua__lapi.o .build/cross-lua/app__lua__lauxlib.o .build/cross-lua/app__lua__lbaselib.o .build/cross-lua/app__lua__lcode.o .build/cross-lua/app__lua__ldblib.o .build/cross-lua/app__lua__ldebug.o .build/cross-lua/app__lua__ldo.o .build/cross-lua/app__lua__ldump.o .build/cross-lua/app__lua__lfunc.o .build/cross-lua/app__lua__lgc.o .build/cross-lua/app__lua__llex.o .build/cross-lua/app__lua__lmathlib.o .build/cross-lua/app__lua__lmem.o .build/cross-lua/app__lua__loadlib.o .build/cross-lua/app__lua__lobject.o .build/cross-lua/app__lua__lopcodes.o .build/cross-lua/app__lua__lparser.o .build/cross-lua/app__lua__lrotable.o .build/cross-lua/app__lua__lstate.o .build/cross-lua/app__lua__lstring.o .build/cross-lua/app__lua__lstrlib.o .build/cross-lua/app__lua__ltable.o .build/cross-lua/app__lua__ltablib.o .build/cross-lua/app__lua__ltm.o .build/cross-lua/app__lua__lundump.o .build/cross-lua/app__lua__lvm.o .build/cross-lua/app__lua__lzio.o .build/cross-lua/app__lua__luac_cross__luac.o .build/cross-lua/app__lua__luac_cross__loslib.o .build/cross-lua/app__lua__luac_cross__print.o .build/cross-lua/app__lua__..__modules__linit.o .build/cross-lua/app__lua__..__libc__c_stdlib.o  -lm
`

Something is missing here? My fault or recent changes in the trunk made it not working?

Thanks

ernitron, 2016-04-12:
I guess this could be related to issue 1107. Please confirm

https://github.com/nodemcu/nodemcu-firmware/issues/1107

TerryE, 2016-04-12:
Which version are you using. The last bit of my latest dev version gives: 
```
[COMPILE] .build/cross-lua/app__lua__luac_cross__print.o
[COMPILE] .build/cross-lua/app__lua__..__modules__linit.o
[COMPILE] .build/cross-lua/app__lua__..__libc__c_stdlib.o
[LINK] luac.cross
[builder] Done building target.
```
Some errors crept in in some changes the where carried into master but have been fixed in dev.

ernitron, 2016-04-14:
I was working on the master branch. 

I downloaded the dev branch and actually it DOES compile luac.cross.

Actually on the dev branch, I cannot compile the firmware but never mind I will use the stable branch.
```

xtensa-lx106-elf-gcc -L/opt/ESP8826-tools/nodemcu-firmware-dev/sdk/esp_iot_sdk_v1.5.1/lib -L/opt/ESP8826-tools/nodemcu-firmware-dev/sdk/esp_iot_sdk_v1.5.1/ld  -Wl,--gc-sections -Wl,-Map=mapfile -nostdlib -T../ld/nodemcu.ld -Wl,@../ld/defsym.rom -Wl,--no-check-sections -Wl,--wrap=_xtos_set_exception_handler -Wl,-static -u ADC_module_selected -u BIT_module_selected -u DHT_module_selected -u FILE_module_selected -u GPIO_module_selected -u I2C_module_selected -u MQTT_module_selected -u NET_module_selected -u NODE_module_selected -u OW_module_selected -u SPI_module_selected -u TMR_module_selected -u UART_module_selected -u WIFI_module_selected -Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lwpa2 -lmain -ljson -lsmartconfig -lssl -lcrypto user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a json/.output/eagle/debug/lib/libjson.a platform/.output/eagle/debug/lib/libplatform.a task/.output/eagle/debug/lib/libtask.a libc/.output/eagle/debug/lib/liblibc.a lua/.output/eagle/debug/lib/liblua.a lwip/.output/eagle/debug/lib/liblwip.a coap/.output/eagle/debug/lib/coap.a mqtt/.output/eagle/debug/lib/mqtt.a u8glib/.output/eagle/debug/lib/u8glib.a ucglib/.output/eagle/debug/lib/ucglib.a smart/.output/eagle/debug/lib/smart.a spiffs/.output/eagle/debug/lib/spiffs.a cjson/.output/eagle/debug/lib/libcjson.a crypto/.output/eagle/debug/lib/libcrypto.a dhtlib/.output/eagle/debug/lib/libdhtlib.a tsl2561/.output/eagle/debug/lib/tsl2561lib.a http/.output/eagle/debug/lib/libhttp.a net/.output/eagle/debug/lib/libnodemcu_net.a modules/.output/eagle/debug/lib/libmodules.a -Wl,--end-group -lm -o .output/eagle/debug/image/eagle.app.v6.out 
/opt/ESP8826-tools/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory `/opt/ESP8826-tools/nodemcu-firmware-dev/app'
make: *** [.subdirs] Error 2

```

TerryE, 2016-04-14:
See #1111.   It's very moot whether `master`is more stable than `dev`.  We've fixed a lot of bug fixes in `dev`.  I personally wouldn't consider using `master`: see #1146.

closed 2016-04-14 by TerryE