nodemcu/nodemcu-firmware #1241: UART Bug

pretcorr, 2016-04-14:
I found a bug that makes the serial data in the UART not transmit back to the port. I send data to the unit via wifi and do not get back the data unless I say how many bytes I need. I got the code working on one unit but cannot get it going on the later ones.

Im using these files

NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: file,gpio,net,node,ow,tmr,uart,wifi
 build built on: 2016-04-13 11:14
 powered by Lua 5.1.4 on SDK 1.4.0

The code I use is

 uart.on("data", 0, function(data)
 conn:send(data)
 end, 0) 

I send 4 bytes then need to get 2 back
I send 4 bytes then 18 comes back
I send 4 bytes then 82 comes back.

Its circular comms with variable bytes being received.

According to the spec the bytes in buffer should be transmitted ie if there is 10 its sent regardless but it does not do that. If I change the value to the value of bytes I expect then it fine. Cannot make it 100 because variable data being send and received.

Thanx!

TerryE, 2016-04-14:
Use a `dev` build, and post your complete code to a gist, so we can review.

pretcorr, 2016-04-14:
Thank you fro responding!

must say the could build service is great!

I used ESplorer to write the code. The implementation is to run serial petrol pump communication through the wifi system at a site and have the pos just connected to the network. It worked on one module but then it stopped I could duplicate the success.

use the build off the could site in dev mode and its doing the same!

NodeMCU custom build by frightanic.com
	branch: dev
	commit: a33f5868551000d6f8e06def540bc5755b2146f0
	SSL: false
	modules: file,gpio,net,node,ow,tmr,uart,wifi
 build 	built on: 2016-04-14 08:15
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

--variables
strresult = '';
dummy = 0;
mycount = 1
txdata = 0
----tag reader pins
TAGONETAG = 2  
--TAGTWOTAG = 1
----tag led pins
TAGTONELED = 5
--TAGTWOLED = 13
--TAGTWOLED = 14
mytagnumber=0
--onboard led
onboardled=4
gpio.mode(onboardled,gpio.OUTPUT)
gpio.write(onboardled,gpio.HIGH) --OFF
--gpio.write(onboardled,gpio.LOW)   --ON
tmr.alarm(0,2000,0,function()end) -- run after a delay//200
--setup wifi
--Number: current mode after setup. 
-- 1 for wifi.STATION; 
-- 2 for wifi.SOFTAP; 
-- 3 for wifi.STATIONAP; 
wifi.setmode(wifi.STATION)
--wifi.setmode(wifi.SOFTAP)
--wifi.setmode(wifi.STATIONAP)
--wifi.sta.config("TotalGlencoe","Lerm7861")
wifi.sta.config("PRETCORR","AA122448FF")
--wifi.sta.connect()
wifi.sta.autoconnect(1)
--wifi.sta.setip({ip="10.0.0.201",netmask="255.255.255.0",gateway="10.0.0.2"})
wifi.sta.setip({ip="192.168.1.201",netmask="255.255.255.0",gateway="192.168.1.1"})

--cfg={}
--cfg.ssid="PRETCORR"
--cfg.pwd="AA122448FF"
--cfg.ip="192.168.1.2011"
--cfg.netmask="255.255.255.0"
--cfg.gateway="192.168.1.1"
--set the settings
--wifi.ap.setip(cfg)
--wifi.ap.config(cfg)
--do the connect
tmr.alarm(1, 1000, 1, function() 
    --if wifi.sta.getip()== nil then 
    --if wifi.sta.status() >> 5 then 
        --0: STATION_IDLE, 
        --1: STATION_CONNECTING, 
        --2: STATION_WRONG_PASSWORD, 
        --3: STATION_NO_AP_FOUND, 
        --4: STATION_CONNECT_FAIL, 
        --5: STATION_GOT_IP.
        mystatus = wifi.sta.status()
[... 296 lines cut ...]
function myservernewnew()
  -- 30s time out for a inactive client
  sv = net.createServer(net.TCP, 30)
  -- server listens on 80, if data received, print data to console and send "hello world" back to caller
  sv:listen(5001, function(c)
    c:on("receive", function(c, pl) 
      print(pl)
    end)
  c:send("hello world")
end)

end
myserverrunning()
--myloopback()
--myservernew()
--myservernewnew()

firmware programmer

C:\Users\Gerhard\Desktop\ESP8266\esp_iot_sdk_v1.4.0\bin\esp_init_data_default.bin 0xFC000
C:\Users\Gerhard\Desktop\ESP8266\esp_iot_sdk_v1.4.0\bin\blank.bin 0x7E000
C:\Users\Gerhard\Desktop\ESP8266\esp_iot_sdk_v1.4.0\bin\blank.bin 0X1FE00
C:\Users\Gerhard\Desktop\ESP8266\LUA Firmware\13042016\nodemcu-dev-8-modules-2016-04-14-08-16-29-float.bin 0x00000

Dont mind the code to be shared! All of it is copied off the web!

pretcorr, 2016-04-14:
Sorry abot the code some parts is commented out! its a work in progress

pretcorr, 2016-04-14:
I run the following baud rates:
4800, 8, N, 1
4800, 8, E, 1
4800, 8, O, 1
9600, 8, N, 1
9600, 8, E, 1
9600, 8, O, 1
14400, 8, N, 1

Are they all supported? I use 9600, 8, N, 1

TerryE, 2016-04-14:
See #719, what we do here is to fix bugs in the firmware; other forums provide general user support. My only Q here is: is there evidence of a bug? We aren't to provide teaching support.  I did ask for a gist rather than an inline post, but I will take a quick look.

pretcorr, 2016-04-14:
Yes, If I use uart.on("data",0, function(data) I get no data back. According to the wiki "if n=0, will receive every char in buffer." If I use this then I get nothing back so yes I consider it a bug. A function that the wiki says works does not work. My code, as send to you, works if I make n=the bytes that I need. 

A previous unit works on this code but subsequent units and firmware upgrades does not work. This is the main key section of the code to send data back to me and because it does this it doesn't work. Isnt that the whole idea of the ESP8266?

pretcorr, 2016-04-14:
Ok read the 719 and understand what it means but I am firmly assured that the new upgrade stopped sending buffer data back

TerryE, 2016-04-14:
The core of your code does the following:
```Lua
do
  tmr.alarm(0,200,0,function()
    global_c = nil
    intrecbyes = 0
    uart.setup(0, 9600, 8, 0, 1, 0)
    local function some_random_stuff_and_set_intrecbyes( conn, payload)
      -- ...
    end
    srv=net.createServer(net.TCP, 60)
    srv:listen(5001,function(conn)
      conn:on("receive",some_random_stuff_and_set_intrecbyes)
      uart.on("data", intrecbyes, function(data)
           conn:send(data)
      end, 0)
    end)
  end)
end
```
The `uart.on()` is executed as part of the listen callback using the `intrecbyes` value at the connection which is typically zero or whatever was left over from the last connection.  The `conn:on("receive")` callback doesn't come into this.  Programming error, not a firmware bug.

closed 2016-04-14 by TerryE

pretcorr, 2016-04-14:
Ok Terry, Thank you for your help! I apologise blaming your code if my code was wrong!

TerryE, 2016-04-14:
No apologies needed :)

ymxcc, 2016-10-28:
@TerryE  

Hi, All!

Now I am facing a very hard problem, even make the whole day confusion on my mind, as folllows:

I want to read CO2 and PM2.5 value from sensors which are connected with UART. But I found that the received data were sometimes right and sometimes wrong, e.g. PM2.5 received data: 22 17 11 0 0 2 245 0 0 2 249 0 0 2 240 2 0 0 3 229
In my code, I just judged the existing "22" in the received data, then the pm2.5 could be calculated using the formulation. However, the data absolutely were wrong, PM2.5 = 757. 
The correct data were: received data: 22 17 11 0 0 0 28 0 0 0 28 0 0 0 25 0 0 0 3 122, the reasonable value is PM2.5 = 28. The bold numbers above made a big difference in some time, especially I downloaded the code into me device with plugging off the USB, making it operation under the battery.

Maybe some guys could help me to find my code where I made a big mistake to make it happen.

function measure(num, switch)

local tem = switch
uart.on("data", num, function(data)
switch = 0
uart.on("data")
uart.alt(0)
uart.setup(0, 115200, 8, uart.PARITY_NONE, uart.STOPBITS_1, 1)

        print("receive from uart:", data)
        print("received data:", string.byte(data, 1, num))
        print("tem", tem)
        if num == 20 then
        pm25_i = 1
        while(pm25_i < num)
        do
            pm25_a = string.byte(data, pm25_i)
            if pm25_a == tonumber('16', 16) then
                pm25_b = tonumber(string.byte(data, pm25_i - 1 + 7), '10') --+ string.byte(data,5) * math.pow(2,8) + string.byte(data,4) * * math.pow(2,16) + string.byte(data,5) * * math.pow(2,24)
                pm25_c = tonumber(string.byte(data, pm25_i - 1 + 6), '10')
                pm25_e = tonumber(string.byte(data, pm25_i - 1 + 5), '10')
                pm25_f = tonumber(string.byte(data, pm25_i - 1 + 4), '10')
                qiaqia_pm25 = pm25_b + bit.lshift(pm25_c, 8) + bit.lshift(pm25_e, 16) + bit.lshift(pm25_f, 24)
                m:valuechange("4", qiaqia_pm25)
                print(qiaqia_pm25)
                break
            end
            pm25_i = pm25_i + 1
        end
        end

        if num == 9 then
        co2_i = 1
        while(co2_i < num)
        do
            co2_a = string.byte(data, co2_i)
            co2_b = string.byte(data, co2_i + 1)
            if co2_a == tonumber('ff', 16) and co2_b == tonumber('86', 16) then
                co2_c = tonumber(string.byte(data, co2_i + 2), '10')
                co2_d = tonumber(string.byte(data, co2_i + 1), '10')
                qiaqia_co2 = bit.lshift(co2_c, 8) + co2_d
                m:valuechange("3", qiaqia_co2)
                print(qiaqia_co2)
                break
            end
            co2_i = co2_i + 1
        end
        end

end, 0)

if switch == 1 then
uart.alt(1)
uart.setup(0, 9600, 8, uart.PARITY_NONE, uart.STOPBITS_1, 0)
pin = 10
gpio.mode(pin,gpio.OUTPUT)
gpio.write(pin,gpio.LOW)
uart.write(0, 0xff, 0x01, 0x86, 0, 0, 0, 0, 0, 0x79)
end

if switch == 2 then
uart.alt(1)
uart.setup(0, 9600, 8, uart.PARITY_NONE, uart.STOPBITS_1, 0)
pin = 10
gpio.mode(pin,gpio.OUTPUT)
gpio.write(pin,gpio.HIGH)
uart.write(0, 0x11, 0x02, 0x0B, 0x01, 0xE1)
end
end
awp = 1
tmr.alarm(1, 2000, 1, function()
if awp == 1 then

measure(9, 1)
awp = 0
elseif awp == 0 then
measure(20, 2)
awp = 1 
end
end)

Thanks so much!