nodemcu/nodemcu-firmware #1244: SPI speed limited by firmware

Starofall, 2016-04-15:
The ucg module proposes to use a clockDiv of 0 - which might works fine for the ILI9341.

`spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, spi.DATABITS_8, 0)`

But in the spi_setup lua binding we find this lines:

```
if (clock_div < 4) {
 // defaulting to 8
 clock_div = 8;
}
```

This defaulting should be removed to enable faster TFT rendering.

devsaurus, 2016-04-15:
Thanks for bringing this up @Starofall.

The current implementation for setting up the clock divider requires that it's not below 4. <4 case could have been clipped to 4 but for backwards compatibility I chose 8 as the default - this was the fixed clock divider back in the times. Goal was to maintain the same SPI clock frequency for old scripts where the invalid 0 is still defined.

The ucg examples date back to the time when the `clock_div` parameter was not effective, thus the 0.

Rationale:

1. You're free to reduce `clock_div` in your script to 4 and run at the maximum possible SPI clock frequency of 20&nbsp;MHz.
2. I'll update the examples with `clock_div` = 8.
3. There's potential for minor improvements in ucg's SPI interface code. I never found the time to raise them up to now.

Starofall, 2016-04-15:
Hmm but I was able to remove the limitation from the firmware and run my display using ucg with clock_div of 2 - and also in the 160Mhz mode. This way it works way faster.

devsaurus, 2016-04-15:
I agree that the clock frequency could be pushed beyond CPU / 4. But without detailed hardware docs I'm feeling uncomfortable to advertise an implementation that's supposed to cover all cases.
If someone comes up with a comprehensive solution that's backed by docs and has been verified for send & receive paths under all SPI modes that would be great. Yet, I don't see this, though I'd also be glad to get it.

devsaurus, 2016-04-18:
> If someone comes up with a comprehensive solution

It seems like Espressif did so with the SDK 1.5.3 release. It contains a fairly complete SPI driver that handles a lot of items that were added to our firmware based on reverse engineering. Among them are clock initialization and CPHA & CPOL configuration.

I'll have a look when time permits if nobody else beats me to that...

TerryE, 2016-04-18:
@devsaurus, Arnim so you're down for the SPI review 😉

djphoenix, 2017-01-19:
I see there is a PR that fixed this.
@devsaurus @marcelstoer was you forget to close referenced issue, or there are something to keep tracked?

devsaurus, 2017-01-19:
Got lost in space and time. Thanks for digging this up.

closed 2017-01-19 by devsaurus