nodemcu/nodemcu-firmware #1245: pwm and gpio module interferences

eku, 2016-04-16:
I'm faced with a problem using the latest HEAD of the dev branch on an ESP-3.

Pin 5,6, and 7 are used for PWM. After power on pin 5,.6,7 show a 2,6V level. After running the following commands
```
pwm.setup(5,500,0)
pwm.setup(6,500,0)
pwm.setup(7,500,0)
```
pin 5 and 7 show a 0,0V level, but pin 6 is at 0,5V. If I run
```
gpio.mode(6,gpio.OUTPUT)
gpio.write(6, gpio.LOW)
```
it dropes down to 0,0V. After that I run ```node.restart()```  and the pwm channel works as expected.

Since GPIO is not required for my use case, it is not configured in (unless I'm hunting a bug). But that pin 6 seams so get some suspicous initialisation.

marcelstoer, 2016-04-17:
GPIO initialization is beyond our control, the firmware could but does not interfere with it. Hence, you need to do that in your own code.

closed 2016-04-17 by marcelstoer

eku, 2016-04-17:
How should one do this in Liua for the PWM module if the GPIO module is not compiled in?
I'm pretty sure that this worked for me with an older version of the ```dev``` branch.

marcelstoer, 2016-04-17:
> How should one do this in Lua for the PWM module if the GPIO module is not compiled in?

You can't.

It's been discussed a few times before, see e.g. #897.

eku, 2016-04-17:
@marcelstoer Ok, got the point. But for my understanding the PWM module has to configure the GPIO line on ```pwm.setup```. If it doesn't, how should this ever work?