nodemcu/nodemcu-firmware #1246: Negative temp handling in DHT module

theopensourcerer, 2016-04-17:
I have an ESP8266-12F connected a DHT22 in my Polytunnel. It works fine when the temps are higher than 0'C but I have just discovered a possible issue with negative temperatures; it's first time the temps have dropped that low!

I am running a recent build of the master branch of nodemcu from the automatic builder at nodemcu-build.com/

    nodemcu-master-10-modules-2016-03-26-11-41-01-integer.bin

Here's the basic code I am using to assemble the data read from my DHT22 sensor:

    dht = require("dht")
    status,temp,humi,temp_decimial,humi_decimial = dht.read(readPin)
        if( status == dht.OK ) then
            temperature = temp .. "." .. (temp_decimial/100)
            humidity = humi .. "." .. (humi_decimial/100)
            print("Temperature: " .. temperature .. " deg C")
            print("Humidity: " .. humidity .. "%")

This data is assembled and then set to a mosquito broker.

When the temperature dropped last night the broker received a payload for the temperature like:

Temp: 0.-1, 0.-2 etc. (Note the negative symbol is on the wrong side of the decimal point)

The was indicating that the temperature was -0.1 or -0.2'C

Is this a bug or am I simply not formatting the results correctly? I am going to guess that if the temperature went to -1.0'C or more then the DHT module would probably return the right format for the integer component, but I have no way of testing it... And would the decimal part also have a negative prefix?

TIA

FrankX0, 2016-04-17:
This is correct behavior. First of all: 0 cannot be negative. If you would display (temp*100+temp_decimal) you would get the correct answer. Below -1 you will be facing the same issue again: -1.1 will become -1.-1 in your case. In my opinion you can either use my first example (temperature multiplied by 100) or work your way around this using some if-statements when temperatures drop below 0. Hope this helps.

theopensourcerer, 2016-04-17:
Thanks for the swift reply @FrankX0 I thought it might be me ;-)

closed 2016-04-17 by marcelstoer