nodemcu/nodemcu-firmware #1250: enduser_setup should use mDNS if built-in

unhold, 2016-04-18:
That would make it more useful and should be easy to implement.

I'm thinking about implementing this, so please tell me if you think it makes sense and if somebody is already working on it.

jmattsson, 2016-05-05:
What scenarios do you see where mDNS would be helpful? When the EUS is running it's on its own network and is a largely a captive portal which not only receives all regular DNS requests, but also points everything asked for back at its own IP address. What would mDNS provide that isn't already covered by this?

jmattsson, 2016-05-05:
Also, tagging @jfollas and @robertfoss as they're the other main EUS authors.

unhold, 2016-05-05:
So you imply that a DNS server is already implemented in EUS, and it should reply with it's own IP to all requests? That was not working for me when I tried it then. I had to access it by it's IP address. Will have another look at it. A big thumbs up if it's already implemented! Maybe you could the info from your post to the documentation.

Another reason could be code sharing.

robertfoss, 2016-05-05:
@unhold EUS implements a tiny DNS server that replies with all requests with its own IP. If it isn't working, please help us debug it. Do you have a reproducible failure case?

FrankX0, 2016-05-19:
DNS works fine in EUS. So no need for mDNS.
So close this issue?

unhold, 2016-05-19:
I could not reproduce my issues with the latest dev version.

Only caveat: The HTTP server only serves "GET /". So "anydomain.org" will bring up the page, but "anydomain.org/previously/opened" will not work.

closed 2016-05-19 by unhold

jmattsson, 2016-05-19:
Yeah, that is a bit unfortunate. We did try having it respond to any sub-page as well, but when we did that it interfered/triggered various captive-portal-auto-handling mechanisms on Android/iPhone/WinPhone and the resulting user experience was even worse than what we currently have.

The more annoying issue that I've found is that devices like to keep their DNS entries cached across networks, so even a open http://example.com/ page might not redirect because the browser doesn't bother re-querying DNS.

unhold, 2016-05-20:
@jmattsson thank you for the info, I'm currently looking at captive portal mechanisms.