nodemcu/nodemcu-firmware #1254: Security issue: oversized CoAP hostnames can trigger a stack overflow in coap_request() and lead to remote code execution

jsandin, 2016-04-18:
### Impact:

Lua code that allows untrusted clients to specify the uri for a coap request is susceptible to remote code execution.  A malicious client can send a specially crafted coap uri containing an oversized host that overwrites the saved a0 register on the stack and causes execution to redirect to an address of the attacker's choosing when coap_request() returns.

This does not affect nodeMCU by default - only programs that allow clients to control the coap uri are affected.  The following (toy) example program is an example of a small Lua program that exposes this vulnerability to remote attackers:

`cs = coap.Client(); s=net.createServer(net.UDP); s:on("receive",function(s,c) cs:get("coap://"..c) end); s:listen(8889)`

If you paste this into nodeMCU and send an oversized string of As using netcat to port 8889 on the device, you can trigger a crash.

### Description:

The coap_request() function allocates a 32-byte 'host' buffer on the stack, and copies the host portion of a coap uri (constructed using coap_new_uri()) into this buffer, even if the size of the host portion of the uri exceeds 32 bytes.  An attacker that sends a uri containing an oversized hostname will cause a stack overflow that overwrites other variables on the stack as well as saved registers a12, a13, a14, a15, and a0.  When the function returns, it will used this modified saved a0 value as the target address to execute, allowing the attacker to redirect execution to an address of their choosing.

### Recommendation:

Either reject uri strings that contain hostnames exceeding 32 bytes, or allocate sufficient memory to store the hostname before performing the memcpy on line 357 of coap.c.

Enhance coap_new_uri() to perform further validation of CoAP uris and reject malformed uris.

### Technical details:

from nodemcu-firmware/app/modules/coap.c - this vulnerability is present in both the current development branch and 0.9.6 release:

```C
313 static int coap_request( lua_State* L, coap_method_t m )
314 {
315   struct espconn *pesp_conn = NULL;
316   lcoap_userdata *cud;
317   int stack = 1;
...
326   stack++;
327   pesp_conn = cud->pesp_conn;
328   ip_addr_t ipaddr;
329   uint8_t host[32];
...
348   coap_uri_t *uri = coap_new_uri(url, l);   // should call free(uri) somewhere
349   if (uri == NULL)
350     return luaL_error( L, "uri wrong format." );
351
352   pesp_conn->proto.udp->remote_port = uri->port;
353   NODE_DBG("UDP port is set: %d.\n", uri->port);
354   pesp_conn->proto.udp->local_port = espconn_port();
355
356   if(uri->host.length){
357     c_memcpy(host, uri->host.s, uri->host.length);
358     host[uri->host.length] = '\0';
```

Here, 'url' is the string passed into cs:get(), and it is first passed to coap_new_uri() on line 348 before being copied into the fixed-size stack-allocated 'host' buffer on line 357, resulting in a vulnerability.

Practical exploitation is possible.  The overflow on 357 trashes lots of other stack-allocated variables (including pdu, t, m, uri, payload, among others), so a maliciously crafted request needs to include workable values for many of these variables so that the function can return on line 428.  coap_new_uri() imposes some constraints on the contents of the url, but by placing the payload in brackets (indicating an ipv6 IP address) arbitrary data (including null bytes and 0x3f) can be written to the stack, and workable values can be supplied that trigger this vulnerability and result in remote code execution for the example program given above.  It may be necessary to first crash the device to get the heap into a known state (with predictable addresses) before exploitation is performed.  Further details are available upon request.

marcelstoer, 2016-04-24:
Fixed with #1260, thanks Philip.

closed 2016-04-24 by marcelstoer

jmattsson, 2016-04-26:
Nice report btw, I appreciate the level of detail!

jsandin, 2016-04-27:
My pleasure guys, anytime, thanks for the hard work on nodeMCU and for getting this fixed so quickly.