nodemcu/nodemcu-firmware #1256: configurable (DHCP) hostname?

joysfera, 2016-04-19:
Currently the NodeMCU client in STA mode sends two different hostnames to DHCP server - on the first connection to an AP it sends "ESP_XXXXXX" while on second and subsequent connections it sends "NODE-XXXXX" (where the XXXXX are last five/six digits of the MAC address). This is weird and confusing for the users. It would be better if it always sent the latter - "NODE-XXXXX". Can this be fixed in NodeMCU or is it buried somewhere deep in SDK blobs? I have tried locating it in the source code but couldn't find neither "ESP_" nor "NODE-".

Ideally, this hostname would be user configurable. Since SDK 1.4 there's a call "wifi_station_set_hostname" - could NodeMCU please make this available when configuring STA mode? That would solve it in a very elegant way and programmers could give meaningful names to the NodeMCU units.

Please note that while mDNS might seem as a better fix, I have just found out that it doesn't work in Microsoft systems at all. This hostname via DHCP works in all operating systems (if the DNS is properly configured, i.e. if the DHCP hands on the name to DNS).

Thanks!

devsaurus, 2016-04-19:
[wifi.sta.sethostname](http://nodemcu.readthedocs.org/en/dev/en/modules/wifi/#wifistasethostname) is probably what you're looking for. It has been added with #878 but isn't yet available in `master`. You'll need a compile from `dev` branch.

joysfera, 2016-04-19:
Thanks! I've been using dev but a little bit old.

closed 2016-04-19 by joysfera

marcelstoer, 2016-04-20:
There's actually two issues here: 
- setting hostname manually ✅ 
- two different hostnames being sent to DHCP server ❎ 

@joysfera do you want to create a separate issue for the second?

joysfera, 2016-04-20:
@marcelstoer #1259