nodemcu/nodemcu-firmware #1258: Executing HTTP request followed by another HTTP request

aadeon, 2016-04-19:
I wanted to execute a HTTP request (using http module) and when a response comes, immediately execute another request. The code is more or less is like this:

```
url = "http://google.com"

function firstRequest(callback)
  print("First request, connecting to " .. url)
  http.get(url, nil,
    function(code, data)
        print("First response: " .. code .. ", data: " .. data)
        callback()
    end)
end

function secondRequest()
  print("Second request, connecting to " .. url)
  http.get(url, nil,
    function(code, data)
        print("Second response: " .. code .. ", data: " .. data)
    end)
end

firstRequest(secondRequest)
```

Unfortunately the second request is never executed, in the output I always get:

```
First request, connecting to http://google.com
> First response: 302, data: ....
Second request, connecting to http://google.com
```

The response from the second request never comes back. What am I doing wrong?

I suspect that when the callback function of the first request is being executed it is not possible to "initiate" another HTTP request. If it is the cause, what is the proper way of achieveing such simple sequence of requests?

marcelstoer, 2016-04-20:
> I suspect that when the callback function of the first request is being executed it is not possible to "initiate" another HTTP request.

I never tried myself but why not, have you tried? This all being asynchronous and event-driven I'd expect that to work just fine.

aadeon, 2016-04-21:
There is output of the script I attached, so don't you think it's very likely that I tried? ;)

marcelstoer, 2016-04-21:
Grrr... mea culpa, sorry about that. I didn't study your example carefully enough and particularly had missed the last line (at that time I didn't think it was a complete example). Pity this seems to fall under #719 and should be discussed elsewhere.

closed 2016-04-21 by marcelstoer

aadeon, 2016-05-02:
Hi Marcel,

Thanks for your hint, it works!
I'm just wondering what is the explanation of such behaviour? I would like to avoid such pitfalls in the future...

aadeon, 2016-05-02:
Another question: after flashing custom firmware build with http://nodemcu-build.com/ (*including* `node` module), `node.task` seems to be nil. I used `dev` branch. Has it been removed or any special module has to be selected to make it work?