nodemcu/nodemcu-firmware #1259: ESP_ and NODE- hostnames (if no hostname is set)

joysfera, 2016-04-20:
Currently the NodeMCU client in STA mode sends two different hostnames to DHCP server - on the very first connection to an AP it sends "ESP_XXXXXX" while on second and subsequent connections it sends "NODE-XXXXX" (where the XXXXX are last five/six digits of its MAC address). This is weird and confusing for the users. It would be better if it always sent the latter - "NODE-XXXXX" - especially when the underscore is not allowed character in a hostname for a DHCP, as I've heard.

The "NODE-XXXXX" comes from #878 and @dnc40085 writes there that he sets it in luaopen_wifi but for some reason the "ESP_XXXXXX" gets in for the first connection.

marcelstoer, 2016-04-20:
Thanks for the detailed analysis.

dnc40085, 2016-04-20:
Maybe the code needs to be moved to `user_main.c` so the hostname can be configured earlier

TerryE, 2016-04-21:
You can always set `WIFI_STA_HOSTNAME` in your build, but IIRC the underlying issue is that the Espressif code sets the DHCP hostname during its startup to the format with the _ in it and this isn't valid under the RFC and so isn't C accepted by some router firmware versions. Moving the code won't really help, IMO

marcelstoer, 2016-04-21:
In #875 Arnim suggest to file a bug with Espressif due to the use of the _ but I didn't find one. So, it likely wasn't reported yet.

devsaurus, 2016-04-21:
Last time when I visited this topic I found that `wifi_station_set_hostname` is of volatile nature. Meaning the configured hostname is lost during reset or power-on. If wifi is set to autoconnect then the default name was applied before `wifi_station_set_hostname` became effective. That would explain your observation @joysfera.

As a short term workaround you could try to _disable_ autoconnect once and configure & connect wifi from within your script each time the firmware starts up.

On the long term, Espressif should be notified that the DHCP hostname feature has room for improvement with respect to 2 related issues:
- default hostname is not compliant as discussed in #875 
- it would be beneficial if the hostname provided with `wifi_station_set_hostname` was stored in the non-volatile config area together with SSID, WPA keys, etc.

Any volunteers?

joysfera, 2016-04-21:
@devsaurus my experience is actually different: the first time after configuring the STA mode with new SSID and password the DHCP server gets the "ESP_xxxx" from SDK. Then, after a reboot, when the autoconnect kicks in, DHCP server gets the "NODE-xxxx". I'll try again with wifi.sta.sethostname("myname") in place...

TerryE, 2016-04-22:
I've just downloaded the 1.5.3 SDK and had a look at the espconn / lwip code and diffed it against the 1.5.2 lwip code base.  Most of the changes are in the DHCP code. How significant they are ...

dnc40085, 2016-05-08:
I think if I put the set hostname code in `user_rf_pre_init()` in `user_main.c` it should fix this, since I read somewhere that `user_rf_pre_init()`executes before the WiFi radio is initialized

dnc40085, 2016-05-08:
@joysfera would you be willing to test a [firmware image](http://www.filedropper.com/nodemcudev216c69fhostnametest512k) to see if this fixes the problem?

joysfera, 2016-05-09:
@dnc40085 yes, commit it to dev and let me know, I'll use the wonderful online build service, flash my device and remove the line that sets my own hostname to see what is the default name.

dnc40085, 2016-05-09:
The link in my post above has the firmware image with the potential fix.

The only changes I made were commenting out the task_post_low entry and added `wifi_station_set_hostname` to `user_rf_pre_init` in `user_main.c`

joysfera, 2016-05-09:
Sorry, overlooked it. I'll download the image and check the modules you compiled in. Will report back. The "ESP_xxx" appears only upon the very first connection to an AP. Let's see if I manage to reproduce it.

dnc40085, 2016-05-09:
The firmware image won't give the details that a build from nodemcu-build.com does.

The modules included in the above build are: adc, bit, dht, file, gpio, i2c, mqtt, net, node, ow, spi, tmr, uart and wifi

joysfera, 2016-05-20:
@dnc40085 took me a while to test it because your firmware is so big that my test application didn't compile on it due to insufficient memory. Anyway, the DHCP hostname works now as follows: right after the client connects to an AP for the first time it asks for "ESP_xxxxx" hostname but within less than 1 second some magic happens and the hostname changes to "didItWork". So my conclusion would be that your change to user_rf_pre_init worked, even though the "ESP_xxxx" blinks there for a short moment. I suppose you're going to change the "didItWork" back to "NODE-xxxx"...

dnc40085, 2016-05-26:
@joysfera My apologies, I wasn't sure which modules to include so I went with the default ones.
Thank you very much for testing the firmware, I will submit a PR with the fix.

marcelstoer, 2016-05-27:
@joysfera and @dnc40085 thanks for your efforts!

closed 2016-05-27 by marcelstoer