nodemcu/nodemcu-firmware #1261: coap has problems with timers and GC

pjsg, 2016-04-24:
The following code causes a crash:

```
cs = coap.Client()
cs:get("coap://1.1.1.1/crash")
cs = 0
```
This is because (I think) a timer starts when the get is performed, but, by the time it goes off, the underlying espconn structure has been deleted and this leads to disaster. 

Since I know almost nothing about coap, I have no real idea how to fix this......

NicolSpies, 2016-05-21:
@pjsg Hi P, from one ham to another, apologies for a "non-firmware issue". I see you have been exploring COAP in the last while. I am about to embark on the same route, from the lua side,.. or should I rather leave it for now due to work in progress, specifically related to the server not returning any data from a client GET ?

pjsg, 2016-05-21:
I am not doing anything with COAP -- I came across the above problem when trying to reproduce another problem submitted as an issue. There are lots of possible solutions, but I'm not sure which one is best -- and I don't have a good testing environment to make sure that any fixes do not break anything else.

Feel free to do what you want!

NicolSpies, 2016-05-21:
Thanks for the feedback, I will investigate

closed 2019-06-21 by stale[bot]