nodemcu/nodemcu-firmware #1267: DHCP IP address visible to router, Static IP address not

Foxdesign2016, 2016-04-27:
Using Esplorer IDE to program the ESP8266-01 --> letting the router assign an ip automatically (using wifi.sta.config("SSID","PW") works fine (I can use it as a web server and see it on my routers list of devices) but changing it to a static IP address (using wifi.sta.setip({ip="",netmask="",gateway=IPROUTER}) ) can still access the web server with the static IP fine, but the router doesn't see the esp8266-01 anymore. Any ideas?

karrots, 2016-04-27:
Depends on where you are "seeing" the router when it's in DHCP mode. It will not show up in the DHCP leases table when a static IP is used. If your router has the ability to show the arp table it may show up there.

This doesn't really sounds like a NodeMCU issue, though. See http://nodemcu-firmware.readthedocs.org/en/dev/en/support/

Foxdesign2016, 2016-04-27:
the router shows all devices on the network on this table(changed phone to static IP and it shows on the router). 
I floated this issue elsewhere and someone commented that the Arduino IDE firmware works( not that I've tested it myself)

dnc40085, 2016-04-27:
Are you having any issues communicating with the esp8266 or is the issue just that it's not showing up in the router's lists?

marcelstoer, 2016-04-28:
This ain't a bug with the firmware -> closing.

closed 2016-04-28 by marcelstoer